fwupd/firmware-dell #110: Latitude 3410 BIOS missing 1.12.0

pjwelsh, 2022-01-31:
latitude+3410
LVFS version  1.10.1
https://fwupd.org/lvfs/search?value=latitude+3410
Dell version 29 Dec 2021,1.12.0, 1.12.0
https://www.dell.com/support/home/en-us/product-support/product/latitude-14-3410-laptop/drivers

pjwelsh, 2022-02-20:
@dell-client-linux please update to latest dell "Urgent".

pjwelsh, 2022-03-09:
Someone from Dell should really be paying attention to missing updates. Anyone representing Dell looking at these?

hughsie, 2022-03-09:
@dell-client-linux is forwarding them to the internal teams, I hope.

sunnycrown83, 2022-03-10:
@pjwelsh Expect to be ready soon, Dell internal is working on that. And a new process in under setup and will keep firmware updated in the near future.

sunnycrown83, 2022-03-10:
@hughsie Does the LVFS server work now? ODM told me "Internal Sever Error" shown after he saved description information.

sunnycrown83, 2022-03-10:
> @hughsie Does the LVFS server work now? ODM told me "Internal Sever Error" shown after he saved description information.

Please ignore this comment, 1.13.0 has been there.

pjwelsh, 2022-03-10:
The newest (1.13.0) does appear on the site and is showing in fwupdmgr. Thank you for your actions. closing

pjwelsh, 2022-03-10:
Just a quick note to make sure that the significant list of outdated BIOS/firmware issues are going to also be addressed?
I will check my script to see which are currently outstanding and comment/close accordingly.

closed 2022-03-10 by pjwelsh