fwupd/firmware-dell #111: Precision 7920/7820 Tower System BIOS no update since 2.6.3 (current 2.18.0)

kuehnerm, 2022-02-02:
The leatest version of the Precision 7920/7820 Tower System Firmware is from 28 May 2020 and since then many new versions including both functional fixes but also many many urgent security fixes have been released.

Firmware in LVFS (00.2.6.3):
https://fwupd.org/lvfs/devices/com.dell.uefia93d4a5b.firmware

Link to Dell firmware:
https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=7m2ny&oscode=wt64a&productcode=precision-7920-workstation&lwp=rt

Installing from .exe works:
`./contrib/firmware_packager/install_dell_bios_exe.py ~/7X20T_2.18.0.exe`

After reboot
`fwupdmgr get-updates
No upgrades for Precision 7920 Tower System Firmware, current is 0.2.18.0: 0.2.6.3=older, 0.2.5.0=older, 0.2.4.1=older, 0.2.3.1=older, 0.2.1.4=older`

This was tested on a Precision 7920.

kuehnerm, 2022-05-11:
Bios 2.20.0 was released containing not published, urgent CVE fixes. Is there anything I can do to help?

CragW, 2022-05-25:
@kuehnerm thanks for reporting this issue, version 2.23.1 is now available.

closed 2022-05-25 by CragW