fwupd/firmware-dell #115: WD19TB fan cycling with 01.00.24.01

KenMacD, 2022-04-07:
Hello, I had to downgrade my Dell dock after installing the latest firmware update because it caused the docks fan to continually cycle on and off at around one second intervals. Reverting to `01.00.04.01` has fixed the issue.

The update/downgrade did replace a few different firmwares for chips in the dock. Other than the fan the update appeared to work fine.

CragW, 2022-04-08:
@KenMacD, a few questions:
1. What model is the host system?
2. Any downstream device connected to WD19TB?
3. When issue happens e.g. `1.0.24`, can you please post the output from `get-devices`?
4. Have you contacted dell service support to reporting this issue?

KenMacD, 2022-04-08:
@CragW Thanks for looking at this. In attempting to answer your questions I've re-updated the dock. Oddly I'm not seeing the same fan behaviour this time. I've followed the same process of applying the update, disconnecting from the dock, and then reconnecting some time later. I also did a system update since initially reporting this, so perhaps that changed something. Either way, feel free to close this and I'll re-open if I see it happen again.

To answer what I can:

> What model is the host system?

Latitude 7420

> Any downstream device connected to WD19TB?

Umm... yes, all the things. Let's see, at the time I think it was the following:

 * Two Dell P2418D monitors connected over DP (I think)
 * a CX31993 based usb audio dac
 * a 9354:34068 wireless mouse receiver
 * one of the monitors USB hubs
   * keyboard that calls itself a 1096:0284

> When issue happens...

Can't reproduce it, but I'll attach the `get-devices` and `get-history` in the hope that it might be useful if others report a similar issue.

[devices.txt](https://github.com/fwupd/firmware-dell/files/8452364/devices.txt)
[history.txt](https://github.com/fwupd/firmware-dell/files/8452365/history.txt)

> Have you contacted dell service support to reporting this issue?

No. I was waiting to see what happened here first.

CragW, 2022-04-12:
The devices.txt indicates that you have the latest firmware bundle 1.0.24 installed, good to know everything works fine now. Please reopen if required.