fwupd/firmware-dell #122: Dell Optiplex 3050 missing urgent BIOS update 1.20.0

pjwelsh, 2022-04-26:
LVFS version  1.19.0
https://fwupd.org/lvfs/search?value=optiplex+3050
Dell version 20 Apr 2022,1.20.0, 1.20.0
https://www.dell.com/support/home/en-us/product-support/product/optiplex-3050-desktop/drivers

pjwelsh, 2022-05-16:
Any word on update time frame for this? Thanks.

pjwelsh, 2022-05-19:
@dell-client-linux just a nudge to see if this BIOS can be updated, please.

pjwelsh, 2022-06-03:
@superm1  or @dell-client-linux just a nudge to see if this BIOS can be updated, please.

superm1, 2022-06-03:
Just FYI I'm not with Dell, I just used to work there. So I won't be able to help get things like getting firmware posted. The GitHub tag you did for the client team is appropriate.

pjwelsh, 2022-06-03:
Apologies for the @superm1 addition... correct request should have been @sunnycrown83

hugh712, 2022-06-06:
@pjwelsh 

Hi,

The BIOS team has been informed, I will provide news for further feedback.

pjwelsh, 2022-06-09:
I see the updated 1.20.0 BIOS is now published. However, the unit I tried testing the update on does not yet see that update even after a "fwupdmgr --force --assume-yes refresh". I will wait to see if this is just a LVFS content dissemination issue and try on Friday or Monday.