fwupd/firmware-dell #123: Dell Latitude 3410 missing urgent BIOS update 1.14.1

pjwelsh, 2022-04-26:
LVFS version  1.13.0
https://fwupd.org/lvfs/search?value=latitude+3410
Dell version 18 Apr 2022,1.14.1, 1.14.1
https://www.dell.com/support/home/en-us/product-support/product/latitude-14-3410-laptop/drivers

pjwelsh, 2022-05-16:
Dell BIOS version is now 1.15.0. Please update.

pjwelsh, 2022-05-19:
@dell-client-linux just a nudge to see if this BIOS can be updated, please.

chinyakao, 2022-06-06:
Hi @pjwelsh, We are processing the issue. we will update it ASAP. Thank you for informing us.

pjwelsh, 2022-06-14:
@chinyakao  @dell-client-linux any ETA on this Latitude 3410 missing BIOS.
Thanks

pjwelsh, 2022-06-20:
@chinyakao @dell-client-linux any ETA on this Latitude 3410 missing BIOS.

Looks like you turned around those Opliplex x060 and x080 BIOS updates from last week very quickly, thank you.

pjwelsh, 2022-06-30:
@chinyakao @dell-client-linux any ETA on this Latitude 3410 missing BIOS.

pjwelsh, 2022-07-07:
07 Jul 2022,"1.16.0, 1.16.0",Urgent,Latitude_3410_3510_1.16.0.exe,"This package contains the Dell system BIOS update. BIOS is a firmware package that is embedded on a small memory chip on the system board. It controls the keyboard, monitor, disk drives, and other devices. This update addresses Common Vulnerabilities and Exposures (CVE). Common Vulnerabilities and Exposures (CVE) is a list of security vulnerabilities and exposures that are publicly disclosed."
https://www.dell.com/support/home/en-us/product-support/product/latitude-14-3410-laptop/drivers

@chinyakao @dell-client-linux please update this BIOS. This is now the third "Urgent" update missed.

pjwelsh, 2022-07-14:
I'm not sure who updated the BIOS for this unit, but thank you. It may be the case that whomever it was did not know of this ticket as a comment of completion was not made... regardless, these BIOS omissions  have very recently been corrected.
closing.