fwupd/firmware-dell #124: Dell XPS 13 9310 2-in-1: System firmware 2.8.1 and 2.9.0 missing on LVFS

paulmenzel, 2022-04-29:
The latest available version on [LVFS is 2.7.0](https://fwupd.org/lvfs/devices/com.dell.uefia472a89c.firmware) from January 11th, 2022, [while version 2.8.1 from March 14th, 2022 and 2.9.0 from April 26th, 2022 are available](https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=3yc92&oscode=wt64a&productcode=xps-13-9310-2-in-1-laptop).

CragW, 2022-05-09:
XPS 9310 2-in-1 doesn't support Ubuntu hence no firmware uploads for fwupd/LVFS.
https://www.dell.com/support/home/en-us/product-support/product/xps-13-9310-2-in-1-laptop/drivers

The latest system firmware for XPS 9310 is 3.6.0:
https://fwupd.org/lvfs/devices/com.dell.uefiecfae66b.firmware

Same as Windows' offering:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=10w50&oscode=ubt20&productcode=xps-13-9310-laptop

closed 2022-05-09 by CragW

paulmenzel, 2022-05-09:
> XPS 9310 2-in-1 doesn't support Ubuntu hence no firmware uploads for fwupd/LVFS. https://www.dell.com/support/home/en-us/product-support/product/xps-13-9310-2-in-1-laptop/drivers

So, why are some older versions available?

@hughsie, it’d be great, if you could talk to Dell about this.

> The latest system firmware for XPS 9310 is 3.6.0: https://fwupd.org/lvfs/devices/com.dell.uefiecfae66b.firmware

Well, that’s a different device.

> Same as Windows' offering: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=10w50&oscode=ubt20&productcode=xps-13-9310-laptop

Same different device.

Please re-open this issue.

CragW, 2022-07-22:
Sorry for the delayed response. The policy for system firmware release is primarily on the compatible systems at launch, for alternative install see if issue #4788 helps but it takes own compatibility risk.

pjwelsh, 2022-08-09:
@CragW - I am also confused by the policy statement.

In fact, there is realistically no "Ubuntu" issue either as I see it.
Dell's commitment is to provide BIOS updates for models of systems to fwupd. fwupd's is where the Ubuntu support would be needed, not Dell. @paulmenzel has pointed this out by showing the existing BIOS versions for this model exists.

@dell-client-linux Please continue the commitment to maintain existing Dell "supported" BIOS versions consistent with what the official Dell support website has here on fwupd. This Dell XPS 13 9310 2-in-1 certainly looks to fall into this category.