fwupd/firmware-dell #126: fwupd web and fwupdmgr BIOS issue for Optiplex 3040

pjwelsh, 2022-05-17:
There is something wrong with the path to get the BIOS *web page* and actually via the fwupdmgr:
https://fwupd.org/lvfs/search?value=optiplex+3040 will show a Optiplex 3040 description. However, it produces a "Dell Optiplex 3046" page at https://fwupd.org/lvfs/devices/com.dell.uefid63450d6.firmware that contains mixed 3040 and 3046 BIOS entries.
Also, fwupdmgr does not see the 3040 updates:
INFO: fwupdmgr --force --assume-yes refresh exit status = 0
==begin ='fwupdmgr --assume-yes get-updates'
Specified firmware is already installed '5.81.2.1'
Specified firmware is older than installed '0.1.16.0 < 0.1.17.1'

Please help get both the web page and actual BIOS updates corrected.

CragW, 2022-05-18:
Optiplex 3040 and 3046 are sharing the same appstream ID, in this case extra requirement on CHID has been added to the firmware component individually hence you probably see the differentiation on physical hardware.

closed 2022-05-18 by CragW

pjwelsh, 2022-05-19:
There may be some issue with CentOS 7 based fwupdmgr compared to CentOS 8 fwupdmgr or something else?
CentOS 7 based systems do *not* see the Optiplex 3040 1.19.1 BIOS and *only* the 1.17.1 (but shows as created 5/17/22):
fwupdmgr get-devices
OptiPlex 3040 TPM 1.2
  DeviceId:             17767aff36e478e69d7b885646f23e0b939e6ab8
  Guid:                 bc4c153b-7398-538e-b182-ed3da1108b69
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              5.81.2.1
  Icon:                 computer
  Created:              2022-05-17

OptiPlex 3040 System Firmware
  DeviceId:             b68427c169463946c6951df617c135d8d5d16efd
  Guid:                 d63450d6-d611-48ac-8f3b-8d29bad80248
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.17.1
  VersionLowest:        0.1.17.1
  Icon:                 computer
  Created:              2022-05-17

Any idea why this would be the case? I've not noticed this kind of issue with any BIOS update in the past.

pjwelsh, 2022-05-19:
As a couple more notes:
Yes, "fwupdmgr --force --assume-yes refresh" has been run on all of the C7 based 3040 systems and all have the same "issue".
I can't seem to re-"open" this issue. Should I create a new?

hughsie, 2022-05-19:
@CragW are you sure it's the correct CHID? I only see a very small number of downloads.

CragW, 2022-05-20:
@pjwelsh can you share hwids on your 3040 system? `$ sudo fwupdtool hwids`

pjwelsh, 2022-05-20:
fwupd-1.0.8-5.el7.x86_64 apparently does not like the hwids command option:
/usr/libexec/fwupd/fwupdtool hwids
Command not found

Can I get that info some other way for this CentOS 7 based system?

Here is the info from a (bios updating ok) CentOS 8 system:
fwupdtool hwids
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.19.1
BiosMajorRelease: 1
BiosMinorRelease: 19
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Dell Inc.
Family: OptiPlex
ProductName: OptiPlex 3040
ProductSku: 6BB
EnclosureKind: 3
BaseboardManufacturer: Dell Inc.
BaseboardProduct: 5XGC8

Hardware IDs
------------
{84839a1d-e741-53b1-a64b-e15196e4d334}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{849c1c02-be4b-5411-a725-26d5acb9f55b}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{84201df4-8924-53d9-b7af-88b1682d9040}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{fb2e061f-f40d-507e-b591-e6a967768548}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{32c0b785-ae33-59da-bd07-35d3f20a105d}   <- Manufacturer + Family + ProductName + ProductSku
{10b085a6-4c88-5af5-834b-302a99a3c44d}   <- Manufacturer + Family + ProductName
{018aa51d-e85d-5002-aab2-b8dc5b63dd2c}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{30e89067-f91e-55b4-8e1f-1dc75dba8896}   <- Manufacturer + ProductSku
{c971c2c0-dc4b-5817-ad5f-50cf7fed293b}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{91721c9c-c1b3-5fe2-b191-9d8e520ed52e}   <- Manufacturer + ProductName
{bd685960-bcd7-559e-bd96-b8f2b3c9fe68}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{91251071-a01b-5067-9dca-81de39f555f9}   <- Manufacturer + Family
{529aea4b-4247-5705-89e2-edc2edd08c4e}   <- Manufacturer + EnclosureKind
{8bfe63ff-4d08-5d10-a849-bef8198bf0ce}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer

CragW, 2022-05-20:
`{10b085a6-4c88-5af5-834b-302a99a3c44d} <- Manufacturer + Family + ProductName`

Above CHID in your list is presently in use, specifically in version 1.19.1.
https://fwupd.org/lvfs/devices/com.dell.uefid63450d6.firmware

pjwelsh, 2022-05-23:
I'm not versed enough to understand the full implication of the above statement.
I can say that for some reason, this particular BIOS update is not seen/find-able by any system running CentOS7 for some reason.
The above "working" output was from a correctly functioning CentOS 8 system.

What is different about this BIOS update/addition that makes in incompatible with CentOS fwupd 1.0.8 that not other BIOS update does (that I know)?