fwupd/firmware-dell #127: BIOS update don't apply to (at least) Dell Optiplex 7060 and 5060 if video is not attached

pjwelsh, 2022-05-19:
I have a pattern of non-updating Dell Optiplex (at least) 7060's and 5060's that revolve around not actually being able to fulfill an intended BIOS update if a video cable is *not* physically attached. Basically, fwupdmgr appears to apply the UEFI update pieces, but not actually be able to complete for some reason in a headless mode.

Is this expected or known behavior? I've not been able to search/find any info so far.
How can I further diagnose where the issue is (in headless mode)?
Thanks
PJ

hugh712, 2022-05-20:
@pjwelsh 

Hi, is there any log on the console when you reproduced this issue?
or maybe please add a verbose flag as
`fwupdmgr -v ...` and attach the log to this issue.

And also the journalctl might also help if you can upload your journalctl as 
`sudo journalctl > journalctl.log`

Thank you

pjwelsh, 2022-05-23:
Let me find a 5060 to mess with. I should have information shortly.

pjwelsh, 2022-05-23:
[fwupdmgr-offline-update-no-kvm.txt](https://github.com/fwupd/firmware-dell/files/8755205/fwupdmgr-offline-update-no-kvm.txt)

pjwelsh, 2022-05-23:
[journalctl.log](https://github.com/fwupd/firmware-dell/files/8755235/journalctl.log)

pjwelsh, 2022-05-23:
fwupdmgr-offline-update-no-kvm.txt = "fwupdmgr -v --offline update 2>&1 |tee /tmp/fwupdmgr-offline-update-no-kvm.txt"
journalctl.log was sanitized for names and IPs.

pjwelsh, 2022-05-23:
The BIOS did not update on reboot when no keyboard, video was attached.

The same command (different output file) *did* update when when a keyboard and video was plugged in:
fwupdmgr -v --offline update 2>&1 |tee /tmp/fwupdmgr-offline-update-with-kvm.txt
[fwupdmgr-offline-update-with-kvm.txt](https://github.com/fwupd/firmware-dell/files/8755298/fwupdmgr-offline-update-with-kvm.txt)

I could try to run the command with no keyboard/video and then plug in K/V and *then* reboot to see if the issue related to command execution + keyboard/video or *just* the keyboard/video?
I do have another 5060 to test/check if that helps?

hughsie, 2022-05-23:
@superm1 any old neurons firing on this? Thanks.

pjwelsh, 2022-05-23:
I went ahead and grabbed another 5060 to test.
1. Fresh PXE install of CentOS 8 with KVM attached
2. Disconnect KVM
3. Reboot
4. Run fwupdmgr (fwupdmgr -v --offline update 2>&1 |tee /tmp/fwupdmgr-offline-update-no-kvm-2.txt
5. Attach KVM
6. Reboot
7. BIOS update success!

So, the issue seems squarely to do with the combo of actual Dell BIOS + fwupdmgr BIOS update UEFI process + No keyboard/video
[fwupdmgr-offline-update-no-kvm-2.txt](https://github.com/fwupd/firmware-dell/files/8755864/fwupdmgr-offline-update-no-kvm-2.txt)

I do have another 5060 on standby for testing if needed.

pjwelsh, 2022-05-23:
I can confirm the same BIOS update failure behavior with an Optiplex 3060 as with at least the 5060 and 7060.
1. Fresh PXE install of CentOS 8 with KVM attached
2. Run fwupdmgr (fwupdmgr -v --offline update 2>&1 |tee /tmp/fwupdmgr-offline-update-with-kvm-for-update-but-none-for-reboot.txt
3. *remove* KVM
4. Reboot
5. BIOS update failure!

(KVM = keyboard, video, mouse just to define it somewhere)
[fwupdmgr-offline-update-with-kvm-for-update-but-none-for-reboot.txt](https://github.com/fwupd/firmware-dell/files/8757353/fwupdmgr-offline-update-with-kvm-for-update-but-none-for-reboot.txt)

pjwelsh, 2022-05-23:
Here is the successful BIOS update verbose output with KVM *attached* to the 3060:
fwupdmgr -v --offline update 2>&1 |tee /tmp/fwupdmgr-offline-update-with-kvm-3060-1.txt
[fwupdmgr-offline-update-with-kvm-3060-1.txt](https://github.com/fwupd/firmware-dell/files/8757385/fwupdmgr-offline-update-with-kvm-3060-1.txt)
:

pjwelsh, 2022-05-23:
As a use case note for headless:
1. We have "spare" Dell Optiplex systems at our sites that are online and always up-to-date, but do not have a monitor or keyboard. These are sometimes left in computer racks for just-in-case swap-in replacement and do not have a monitor or keyboard.
2. We also utilize Dell Optiplex systems as a general file server in both a primary and secondary roles (both at every site we have). The file servers will most often not have a monitor or keyboard and most often live in a rack.

superm1, 2022-05-23:
> @superm1 any old neurons firing on this? Thanks.

It sounds like from the last few messages here that the failure is specifically when no display is connected while `UpdateCapsule()` runs from `fwupdx64.efi`.

It probably is that the GOP driver doesn't initialize in headless mode and the firmware flash process has a dependency on it in order to work.  A fix would need to be done in the BIOS on the machine.  

BTW - if my hypothesis is correct I would expect you could also reproduce this with WU capsule updates or via launching the EXE file from Windows over RDP/VNC too.

As a W/A, if you don't have a need to have secure boot turned on you might be able to enable legacy option ROM on these machines which might help.  It should initialize the display to some very basic VESA resolution even with nothing plugged in and can get you around this issue.

superm1, 2022-05-23:
> I will see about testing the legacy boot option. However, as a normal course of BIOS setup/consistency, that options is not enabled for any of our UEFI capable systems.

Yeah, it's just a W/A.  If it works, you might be able to enable to programmatically enable /disable it using dell-wmi-sysman.

pjwelsh, 2022-05-23:
The W/A to disable secure boot and enable legacy boot *does* seem to update the BIOS as predicted on the Optiplex 5060 I tested with.

I'm think I'm understanding that the "fix" will need to be from Dell and a change/correction to the BIOS?

I will attempt to expand my check to newer systems than the x060 series. I have an Optiplex 7080 that is at my disposal to test, but it will need to wait until Wednesday.

superm1, 2022-05-23:
> The W/A to disable secure boot and enable legacy boot does seem to update the BIOS as predicted on the Optiplex 5060 I tested with.

👍 

> I think I'm understanding that the "fix" will need to be from Dell and a change/correction to the BIOS?

Yes, for a proper software solution if it's going to be fixed it would need to be a BIOS fix.  Another "hardware" workaround you can look into is a [dummy plug](https://amzn.to/3wIFGbJ).

hughsie, 2022-05-24:
Thanks @superm1 -- always appreciated.

pjwelsh, 2022-05-24:
Does anyone know how I report the BIOS issue to Dell?
I see that @dell-client-linux was noted earlier. Any input from them?
Any guidance on exactly what should be reported as the issue that will make the most sense for the Dell BIOS peeps? The closest to a working full description of the issue is, "the GOP driver doesn't initialize in headless mode and the firmware flash process has a dependency on it". Should that be enough?

hugh712, 2022-05-24:
@pjwelsh 

Thanks for your information, 
I will help you to report this issue to the Dell BIOS team,
and I will provide the feedback once I have further news.

hugh712, 2022-05-26:
@pjwelsh 

Would you mind providing more information as :
`fwupdmgr get-devices  -v`

hugh712, 2022-05-27:
Feedback from the BIOS team,
this issue should be fixed in BIOS rev. 1.6.1
and the latest version is 1.18.0,
so I'm not sure if the BIOS version is too old will prevent the upgrading or not.

pjwelsh, 2022-05-27:
Sorry, on holiday until next Tuesday (not able to do physical things to check/change), but...
I do not believe the issue was fixed in version 1.6.1...
According to my install script report, one of the Optiplex 5060 tested was at BIOS version *1.13.0* and failed. (it was re-installed and wiped the fwupdmgr history).

According to a current testing 3060, it was at version 1.14.0 and *failed*:
fwupdmgr get-history
OptiPlex 3060
?
??System Firmware:
  ?   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  ?   Previous version:   1.14.0
  ?   Update State:       Success
  ?   Last modified:      2022-05-23 18:53
  ?   GUID:               603297b5-9cc4-49fa-be2a-c14f6e8c1fb1
  ?   Device Flags:       ? Internal device
  ?                       ? Updatable
  ?                       ? System requires external power source
  ?                       ? Supported on remote server
  ?                       ? Needs a reboot after installation
  ?                       ? Device is usable for the duration of the update
  ? 
  ??OptiPlex 3060:
        New version:      1.18.0
        Remote ID:        lvfs
        Release ID:       12947
        Summary:          Firmware for the Dell OptiPlex 3060
        License:          Proprietary
        Size:             11.7 MB
        Created:          2022-03-22
        Urgency:          Low
        Vendor:           Dell
        Description:      
        Fixes & Enhancements - Firmware updates to address security vulnerabilities.

One of the kinda recent test, is a bit more odd: Optiplex 3060 updated by the fwupdmgr seems to have installed a version *1.19.0* at this time (that doesn't exist on the Dell web site -- did they pull an update !!???):
inxi -M
Machine:
  Type: Desktop System: Dell product: OptiPlex 3060 v: N/A serial: D2DPCxx
  Mobo: Dell model: 04Y8V0 v: A02 serial: /D2DPCV2/CNFCW008CC040U/ UEFI: Dell v: 1.19.0
    date: 04/14/2022

fwupdmgr get-devices 
OptiPlex 3060
?
??ADATA SX6000LNP:
?     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
?     Summary:            NVM Express solid state drive
?     Current version:    V9002s72
?     Vendor:             Realtek Semiconductor Co., Ltd. (NVME:0x10EC)
?     Serial Number:      2K5029Q84HAT
?     GUIDs:              227c17b8-29e0-5fc1-af49-a269926d3bc1 ? NVME\VEN_10EC&DEV_5763&SUBSYS_10EC5763&REV_01
?                         2e94b89c-ceda-5e57-a746-db2d1addb802 ? NVME\VEN_10EC&DEV_5763&SUBSYS_10EC5763
?                         b7dbac1e-4f72-55c9-8414-6ea1cd59d3fb ? NVME\VEN_10EC&DEV_5763&REV_01
?                         c119cf33-7b5f-5a49-a621-373832537035 ? NVME\VEN_10EC&DEV_5763
?                         12871ff2-0b79-5bf4-a653-514333c1eee0 ? ADATA SX6000LNP
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Needs a reboot after installation
?                         ? Device is usable for the duration of the update
?   
??System Firmware:
? ?   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
? ?   Summary:            UEFI ESRT device
? ?   Current version:    1.19.0
? ?   Minimum Version:    1.19.0
? ?   Vendor:             Dell (DMI:Dell Inc.)
? ?   Update State:       Success
? ?   GUIDs:              603297b5-9cc4-49fa-be2a-c14f6e8c1fb1
? ?                       230c8b18-8d9b-53ec-838b-6cfc0383493a ? main-system-firmware
? ?                       ba223a54-5b6e-58b7-93a1-b06fcace3b58 ? UEFI\RES_{603297B5-9CC4-49FA-BE2A-C14F6E8C1FB1}
? ?   Device Flags:       ? Internal device
? ?                       ? Updatable
? ?                       ? System requires external power source
? ?                       ? Supported on remote server
? ?                       ? Needs a reboot after installation
? ?                       ? Device is usable for the duration of the update
? ? 
? ??UEFI dbx:
?       Device ID:        362301da643102b9f38477387e2193e57abaa590
?       Summary:          UEFI revocation database
?       Current version:  77
?       Minimum Version:  77
?       Vendor:           UEFI:Linux Foundation
?       Install Duration: 1 second
?       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423 ? UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9
?                         4a6cd2cb-8741-5257-9d1f-89a275dacca7 ? UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
?                         c6682ade-b5ec-57c4-b687-676351208742 ? UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
?                         f8ba2887-9411-5c36-9cee-88995bb39731 ? UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
?       Device Flags:     ? Internal device
?                         ? Updatable
?                         ? Needs a reboot after installation
?                         ? Only version upgrades are allowed
?     
??TPM 2.0:
      Device ID:          a3487e128cf1413519bce8e9a1ab3f5981e61458
      Summary:            UEFI ESRT device
      Current version:    7.2.0.1
      Vendor:             Dell Inc. (PCI:0x1028)
      Update State:       Success
      Update Error:       Updating disabled due to TPM ownership
      GUIDs:              fc72ab1d-dd87-58bb-ac49-1b5d39f5479e ? 085c-2.0
                          ff71992e-52f7-5eea-94ef-883e56e034c6 ? system-tpm
      Device Flags:       ? Internal device
                          ? System requires external power source

pjwelsh, 2022-05-27:
Looking back on my BIOS update tracking, the 1.19.0 version for x060's were added on 5/10/22 and removed around 5/18/22.
==begin changes for optiplex-5060-desktop
10 May 2022,"1.19.0, 1.19.0",Urgent,OptiPlex_5060_1.19.0.exe,"This package contains the Dell system BIOS update. BIOS is a firmware package that is embedded on a small memory chip on the system board. It controls the keyboard, monitor, disk drives, and other devices."
https://www.dell.com/support/home/en-us/product-support/product/optiplex-5060-desktop/drivers
==end changes for optiplex-5060-desktop

==begin changes for optiplex-7060-desktop
10 May 2022,"1.19.0, 1.19.0",Urgent,OptiPlex_7060_1.19.0.exe,"This package contains the Dell system BIOS update. BIOS is a firmware package that is embedded on a small memory chip on the system board. It controls the keyboard, monitor, disk drives, and other devices."
https://www.dell.com/support/home/en-us/product-support/product/optiplex-7060-desktop/drivers
==end changes for optiplex-7060-desktop

==begin changes for optiplex-3060-desktop
10 May 2022,"1.19.0, 1.19.0",Urgent,OptiPlex_3060_1.19.0.exe,"This package contains the Dell system BIOS update. BIOS is a firmware package that is embedded on a small memory chip on the system board. It controls the keyboard, monitor, disk drives, and other devices."
https://www.dell.com/support/home/en-us/product-support/product/optiplex-3060-desktop/drivers
==end changes for optiplex-3060-desktop

hugh712, 2022-05-30:
@pjwelsh 

Since this is a BIOS issue instead of fwupd issue,
so please help report this issue to Dell Product Service Support, thank you,
if there isn't any further question, allow me to close this issue.

[0] https://www.dell.com/support/home/en-us?app=products

closed 2022-05-31 by CragW

pjwelsh, 2022-05-31:
As a followup, it looks like Dell's "fix" was to provide a command line switch to the BIOS EXE called "/novideo" in versions after 1.7.1:
https://www.dell.com/support/kbdoc/en-us/000146859/optiplex-3060-5060-7060-system-doesn-t-upgrade-downgrade-bios-without-a-monitor-connected?lang=en

Since this is now a command line argument to the BIOS install process, should it be part of fwupdmgr options to install the BIOS?

pjwelsh, 2022-06-01:
Is "capsule handler" a fwupdmgr issue or BIOS issue?
Should this ticket be re-opened if fwupdmgr issue or a new ticket created?

pjwelsh, 2022-10-11:
Can anyone offer guidance on where to start with that issue process?
Thanks
PJ

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived.  Cassens