fwupd/firmware-dell #129: XPS13: 9370: missing "Dell XPS 13 9370 System BIOS" - version 1.19.0

colemickens, 2022-05-31:
from: https://www.dell.com/support/home/en-us/product-support/product/xps-13-9370-laptop/drivers
see: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=44dpv&oscode=biosa&productcode=xps-13-9370-laptop

This is not available in fwupd. It's been a week. Thanks!

CragW, 2022-06-01:
Let's see if this release works for you.
https://fwupd.org/lvfs/devices/com.dell.uefi7ceaf7a8.firmware

colemickens, 2022-06-01:
I don't really expect support for this, but I do find it interesting that it seemingly won't apply the update without a display attached. This laptop suffered a fall years ago and I've noticed the BIOS only updates (promptless, note) when there's a screen attached via USB-C->hub->hdmi.

Anyway, once I attached an HDMI device and tried again, it seemed to apply. Thanks for the super fast turnaround!

pjwelsh, 2022-08-01:
Looks like this issue can be closed based on comment from original creator @colemickens

closed 2022-08-05 by CragW