fwupd/firmware-dell #130: How to known why the OptiPlex 7060 System BIOS 1.19.0 was pulled?

rgl, 2022-06-03:
How do I known why the OptiPlex 7060 System BIOS 1.19.0 was pulled?

A couple of weeks ago it was available, but its now gone. Can you help me understand where I should look for this? The page at https://www.dell.com/support/home/en-us/product-support/product/optiplex-7060-desktop/drivers does not seem to have a history.

pjwelsh, 2022-07-14:
They may have pulled due to the quick release of 1.20.0 based on this issue https://www.dell.com/support/kbdoc/en-us/000201396/dsa-2022-186
Hope that helps.

pjwelsh, 2022-08-09:
With the just released 1.21.0 BIOS update for the 7060, you are not likely going to get any notice here as to why the 1.19.0 was pulled :(