fwupd/firmware-dell #134: Dell OptiPlex 5055: Missing system firmware version 1.6.1

paulmenzel, 2022-08-04:
The current version on Dell’s Web site is [1.6.1](https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=hw7pc&oscode=ub16g&productcode=optiplex-5055-ryzen-cpu) released on July 27th, 2022.

[LVFS only contains 1.2.50.](https://fwupd.org/lvfs/search?value=5055)

pjwelsh, 2022-08-08:
FYI: the updated 1.6.1 is showing on fwupd.org now:
https://fwupd.org/lvfs/search?value=5055
May want to very and close if working as expected.

paulmenzel, 2022-08-08:
Thank you for the heads-up. Why the delay of several days though?

pjwelsh, 2022-08-09:
I'm just an end-user/admin and not related to Dell or fwupd - but I find that there can be times that updates can be missed for months and months. So a few days is not so bad ;) I was basically double checking a specific new situation and happened to see yours and checked. I (script) monitor a fair number of Dell models as an admin for my company.

I, like you, have created "missing BIOS" tickets that will *not* get a, "hey, we updated that BIOS" message. However, the BIOS will be updated by someone at Dell... I get the feeling that not all of the "right" Dell people are part of this mailing list at times.

pjwelsh, 2022-08-16:
@paulmenzel : You may want to close this issue now.

paulmenzel, 2022-08-22:
Thank you for the reminder, but I do not want to close the issue, until a Dell employee looked at it, and explains, why the updates were not published right away.

mc-k1, 2024-02-13:
@paulmenzel  - seeing as this ticket has been open for over a year now, would you consider closing it to reduce some of the clutter in the queue?
[disclaimer: I'm not associated with this project. just came across this ticket by chance]

paulmenzel, 2024-02-13:
@mc-k1, thank you for your comment, but please read the whole thread because pinging something. I explained why I do not want to close the issue. I expect that Dell employs people to look at and deal with issues.

mc-k1, 2024-02-14:
@paulmenzel - no worries. It's totally your ticket, so up to you want you want to do with it.
I did read through it, and just thought you might've forgotten about it after a year. So, I gave it a nudge.
I guess that Dell does employ people to deal with issues - but I don't think this repository is Dell-owned. So, you're unlikely to get much attention.

hughsie, 2024-02-14:
Ohh, Dell watches these for sure.