fwupd/firmware-dell #142: Force update with low battery

gabepinheiro, 2022-10-18:
Hi, my battery is not charging and is currently stuck at 2%. To try to solve this problem, I need to update the BIOS, but due to the battery being low, it is not possible to perform the update.

Would there be any way to "force" this and be able to update?

I wanted to take advantage and also request to include the missing BIOS updates for my model: Inspiron 15 3515
LVFS: No update available
https://fwupd.org/lvfs/search?value=inspiron+3515
Dell versions: 1.3.2,  1.4.0, 1.5.0, 1.6.0, 1.7.0, 1.8.0 and 1.9.0
https://www.dell.com/support/home/pt-br/product-support/servicetag/0-c2I2ME15Tis1NXY5eHhHUWdBbzNZZz090/drivers

```
./contrib/firmware_packager/install_dell_bios_exe.py Inspiron_3515_1.3.2.exe
org.freedesktop.DBus.Error.InvalidArgs: No such property “OnlyTrusted”
Installing to System Firmware
Wrote capsule /tmp/tmpw23pebec/firmware.bin
GUID:      3b49e316-45ba-9af2-49e0-6b9f77105d34
HdrSz:     0x1000
Flags:     0x70000
PayloadSz: 0x1393f88
Generated CAB file /tmp/tmpw23pebec/firmware.cab
[                                                  ] 0% idle
FwupdError: Device System Firmware [a45df35ac0e948ee180fe216a5f703f32dda163f] does not currently allow updates: Cannot install update when system battery is not at least 10% (10)
```

hughsie, 2022-10-18:
Try adding `DisabledPlugins=upower` to `/etc/fwupd/daemon.conf `

gabepinheiro, 2022-11-10:
Sorted out

closed 2022-11-10 by gabepinheiro