fwupd/firmware-dell #144: TPM chips does not appear anymore for update on dell 9350

fredoche, 2022-10-22:
Hi, 

Despite issues wih actual updating process, the TPM chip used to appear in the Firmware interface, and fupdmgr get-devices.
It doesnt anymore, yet the devices exists in /dev/tpm0
Before:
[image]

And now:
[image]

```
fwupdmgr get-devices
Dell Inc. XPS 13 9350
│
├─Core™ i7-6560U CPU @ 2.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000f0
│     Fournisseur:        Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         28d7f31c-1663-5b15-a900-94a71cbcd53f ← CPUID\PRO_0&FAM_06&MOD_4E
│                         e4bc7423-9fec-55a6-8f6c-827c40992491 ← CPUID\PRO_0&FAM_06&MOD_4E&STP_3
│     Drapeaux de périphérique:• Périphérique interne
│   
├─Iris Graphics 540:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Version actuelle:   0a
│     Fournisseur:        Intel Corporation (PCI:0x8086)
│     GUIDs:              7b246578-7083-57a2-aad3-f421e8ba9ed0 ← PCI\VEN_8086&DEV_1926
│                         1aa9bc07-3911-5ba9-9ede-7efb78dda7e9 ← PCI\VEN_8086&DEV_1926&REV_0A
│                         37dbff15-684f-5799-ae0d-591edbb218fe ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704
│                         18b3de8b-181c-5dcd-a69d-484d79511211 ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704&REV_0A
│                         c568e060-8b86-5a8c-a86e-56060b784495 ← PCI\VEN_8086&DEV_1926&REV_00
│                         26e22885-57dc-520e-b290-2e727ebff8be ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704&REV_00
│     Drapeaux de périphérique:• Périphérique interne
│                         • Cryptographic hash verification is available
│   
├─PM951 NVMe SAMSUNG 512GB:
│     Device ID:          e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Résumé:             NVM Express solid state drive
│     Version actuelle:   BXV77D0Q
│     Fournisseur:        Samsung (NVME:0x144D)
│     Numéro de série:    S29PNXAH214218
│     GUIDs:              77bb4022-9503-586c-99b7-c8a64502da30 ← NVME\VEN_144D&DEV_A802
│                         cce2386f-6d68-5f85-9f3f-ebead9288696 ← NVME\VEN_144D&DEV_A802&REV_01
│                         484eb66f-4175-5499-8dc9-a8cc651e8450 ← NVME\VEN_144D&DEV_A802&SUBSYS_144DA801
│                         b586caf2-ac7e-5a07-8630-dd289d8fd3f4 ← NVME\VEN_144D&DEV_A802&SUBSYS_144DA801&REV_01
│                         2b05215d-0ecf-5749-a584-012329fb33e6 ← PM951 NVMe SAMSUNG 512GB
│     Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│                         • Le système nécessite une source d'alimentation externe
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Résumé:             UEFI ESRT device
  │   Version actuelle:   1.13.0
  │   Version minimum:    1.13.0
  │   Fournisseur:        Dell (DMI:Dell Inc.)
  │   État de mise à jour:Success
  │   GUIDs:              33773727-8ee7-4d81-9fa0-57e8d889e1fa
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Drapeaux de périphérique:• Périphérique interne
  │                       • Mise à jour possible
  │                       • Le système nécessite une source d'alimentation externe
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Résumé:           UEFI revocation database
        Version actuelle: 217
        Version minimum:  217
        Fournisseur:      UEFI:Linux Foundation
        Durée d'installation:1 seconde
        GUIDs:            c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Drapeaux de périphérique:• Périphérique interne
                          • Mise à jour possible
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
```

```
ls /dev/tpm0 
/dev/tpm0

```

fredoche, 2022-10-22:
I dont have at hand the fwupdmgr version of the top screenshot. It was taken from fedora 37 beta live cd if I recall correctly.

Current version is:

```
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.1
runtime   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    5.19.16-300.fc37.x86_64
runtime   com.dell.libsmbios            2.4
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.4.2
```

hughsie, 2022-10-23:
Can you grab the output of `sudo fwupdtool --plugins --dell get-devices -v` please

fredoche, 2022-10-23:
```
[root@cinderblock fred]# sudo fwupdtool --plugins --dell get-devices -v
11:57:17:0154 FuDebug              Verbose debugging enabled (on console 1)
11:57:17:0155 FuMain               locked /run/lock/fwupdtool
11:57:17:0158 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
11:57:17:0158 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Chargement…              [-                                      ]11:57:17:0175 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
11:57:17:0175 FuCommon             reading /etc/fwupd/daemon.conf with 2224 bytes
11:57:17:0175 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
11:57:17:0176 FuConfig             using autodetected max archive size 4,2 Go
11:57:17:0177 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Chargement…              [ -                                     ]11:57:17:0177 FuRemoteList         loading /usr/share/fwupd/metainfo
11:57:17:0178 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
11:57:17:0178 XbSilo               file: b7da18a7-9720-dbee-fa7f-3c52dfd0b29a, current:b7da18a7-9720-dbee-fa7f-3c52dfd0b29a, cached: (null)
11:57:17:0178 XbSilo               loading silo with file contents
11:57:17:0178 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
11:57:17:0178 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
11:57:17:0178 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
11:57:17:0178 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
11:57:17:0179 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
11:57:17:0179 FuRemoteList         ordering lvfs-testing=lvfs+1
11:57:17:0179 FuRemoteList         ignoring unfound remote fwupd
Chargement…              [                                       ]11:57:17:0179 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:57:17:0179 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:57:17:0179 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:57:17:0179 Jcat                 loaded 1 certificates
11:57:17:0179 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:57:17:0180 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:57:17:0180 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:57:17:0180 Jcat                 loaded 1 certificates
11:57:17:0180 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
11:57:17:0180 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
11:57:17:0180 FuEngine             client certificate now exists: nothing to do
Chargement…              [                                       ]11:57:17:0180 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
11:57:17:0181 FuHistory            got schema version of 8
Chargement…              [*                                      ]11:57:17:0193 FuPlugin             load(/usr/lib64/fwupd-1.8.6/libfu_plugin_modem_manager.so)
11:57:17:0195 FuContext            added udev subsystem watch of platform
11:57:17:0195 FuContext            added udev subsystem watch of mei
11:57:17:0195 FuContext            added udev subsystem watch of block
11:57:17:0195 FuContext            added udev subsystem watch of pci
11:57:17:0195 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
11:57:17:0195 FuPluginDell         Using libsmbios 2.4
11:57:17:0195 FuContext            added udev subsystem watch of i2c
11:57:17:0195 FuContext            added udev subsystem watch of i2c-dev
11:57:17:0195 FuContext            added udev subsystem watch of hidraw
11:57:17:0196 FuContext            added udev subsystem watch of gpio
11:57:17:0196 FuContext            added udev subsystem watch of iommu
11:57:17:0196 FuContext            added udev subsystem watch of msr
11:57:17:0196 FuContext            added udev subsystem watch of mtd
11:57:17:0196 FuContext            added udev subsystem watch of nvme
11:57:17:0198 FuContext            added udev subsystem watch of drm
11:57:17:0199 FuContext            added udev subsystem watch of drm_dp_aux_dev
11:57:17:0199 FuContext            added udev subsystem watch of serio
11:57:17:0199 FuPluginTest         init
11:57:17:0200 FuContext            added udev subsystem watch of thunderbolt
11:57:17:0200 FuContext            added udev subsystem watch of tpm
11:57:17:0200 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
11:57:17:0200 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
11:57:17:0200 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
11:57:17:0200 XbSilo               file: 571fbb78-7115-5d9a-2579-b83161891d17, current:571fbb78-7115-5d9a-2579-b83161891d17, cached: (null)
11:57:17:0200 XbSilo               loading silo with file contents
Chargement…              [*                                      ]11:57:17:0201 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
[... cut ...]
11:57:17:0262 FuBackend            ignoring MEI device /sys/devices/pci0000:00/0000:00:16.0/0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1 as no device file
Chargement…              [-                                      ]
11:57:17:0419 FuCommon             creating path /var/cache/fwupd/motd.d
11:57:17:0420 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
11:57:17:0437 FuMain               No detected devices

11:57:17:0445 FuPluginTest         destroy
```

hughsie, 2022-10-24:
I see the same on the XPS 13 I have here; debugging today.

hughsie, 2022-10-24:
@fredoche can you try the `wip/hughsie/dell-ctx` branch please?

fredoche, 2022-10-24:
I dont have the computer at hand right now. I'll do it as soon as possible.
Thanks

closed 2022-10-24 by hughsie