fwupd/firmware-dell #158: Support for Dell in-between-versions

dcd-arnold, 2023-02-10:


hughsie, 2023-02-10:
@CragW is this missing a firmware requirement perhaps?

CragW, 2023-02-10:
Yeah missing the requirement, I've just added it.

CragW, 2023-02-10:
@dcd-arnold, if you got a chance to `refresh` metadata then the required minimal version should be there. Thanks for reporting this issue.

dcd-arnold, 2023-02-11:
@CragW Tank you for the fast response. I don't have access to the computer this weekend, I will check Monday. However,  I just downloaded the `.cab`-file from LVFS, unpacked it with `cabextract` and found:
```
 41   <requires>
 42     <firmware compare="ge" version="1.10.0"/>
 43   </requires>
```
in the `firmware.metainfo.xml`. I guess that is the change you made?

If I look at the current Bios of the 7470, the [Dell Website for Bios version 1.20.0](https://www.dell.com/support/home/en-us/product-support/servicetag/0-NHc2VWxkaDJtOXhrUnphN3hWU3NFQT090/drivers) states:

> If the current BIOS version is earlier than 1.0.2, ensure that you update the BIOS to version 1.5.0 before updating the BIOS to version 1.20.0.

However, when I look into the `.cab` file [from LVFS](https://fwupd.org/lvfs/devices/com.dell.uefib13b7af6.firmware) for the same Model and Bios version, I don't see the same `requires`. Is that missing, too? 

Thank you very much and have a nice weekend,
dcd-arnold

dcd-arnold, 2023-02-13:
@CragW I did a `fwupdmgr refresh --force` and it shows the latest version again:
```
 $ fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.5.3 to 1.24.0?                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues: Firmware updates to 2022     ║
║ Dec platform update.                                                         ║
║                                                                              ║
║ OptiPlex 7460 AIO must remain plugged into a power source for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
```
Maybe I have the package cached?

CragW, 2023-02-13:
Yup, you have the update candidate listed on your cached metadata. If you proceed to update then the download starts.

dcd-arnold, 2023-02-13:
@CragW please note, that it now offers `1.24.0` and not `1.25.0`. Maybe because the device fulfills the requirements for `1.24.0` version?

CragW, 2023-02-15:
The requirement have just been added on the mentioned releases for Optiplex 7460 and Optiplex 7470.

dcd-arnold, 2023-02-15:
I can verify that upgrades on the 7460 work as expected: I am offered `1.10.0`. I have checke the Bios version `1.7.1` for the 7470, and the requirement is set. Thank you very much!

dcd-arnold, 2023-02-27:
Hi,

I have tested a 7470 now and `fwupd` crashes with:
```
...
11:05:25:0824 FuEngine             1908 components now in silo
11:05:25:0853 FuEngine             requirement 1.4.6 ge 1.7.9 -> org.freedesktop.fwupd passed
11:05:25:0856 FuDevice             changing verfmt for a45df35ac0e948ee180fe216a5f703f32dda163f: number->dell-bios
11:05:25:0856 FuDevice             changing version for a45df35ac0e948ee180fe216a5f703f32dda163f: 70656->1.20.0
11:05:25:0856 FuDevice             changing version lowest for a45df35ac0e948ee180fe216a5f703f32dda163f: 70656->1.20.0
11:05:25:0859 GLib                 g_strsplit: assertion 'string != NULL' failed
Segmentation fault
```

So I have spun up another 7470 and got the same error. If I remove [the local metadata](https://github.com/fwupd/fwupd#local-metadata) with:
```
rm -rf /var/lib/fwupd/*
rm -rf /usr/share/fwupd/*
```

I can run `fwupd` until I run `fwupdmgr refresh`. I assume this is a regression, hence I am posting this here.

I can not find a `cab` file for version `1.4.6` nor `1.7.9`. Within the `cab` file for `1.20.0`, I cannot find any mention of `a45df35ac0e948ee180fe216a5f703f32dda163f` in `/var/lib/fwupd/` or `/usr/share/fwupd/`. 

Where do I go from here?

hughsie, 2023-02-27:
> 11:05:25:0859 GLib                 g_strsplit: assertion 'string != NULL' failed

Super uncool. Can you get a backtrace please? i.e. something like `sudo gdb --args /usr/libexec/fwupd/fwupd --verbose` and then when it crashes do `bt` and then dump the output here.

dcd-arnold, 2023-02-28:
Hi @hughsie,
this is the trace: [gdb.txt](https://github.com/fwupd/firmware-dell/files/10850895/gdb.txt). Anything I can do on my side to help.

hughsie, 2023-02-28:
@dcd-arnold what's the output for `sudo fwupdtool get-devices -v --plugins uefi-capsule` please

hughsie, 2023-02-28:
Look what I spy! `<requires><firmware compare="ge" version="1.5.0"/><client/></requires>` -- that looks like a LVFS website issue as the root cause. Debugging now.

dcd-arnold, 2023-02-28:
> fwupdtool get-devices -v --plugins uefi-capsule

It looks very similar to the error above:
[fwupdtool.txt](https://github.com/fwupd/firmware-dell/files/10851222/fwupdtool.txt)

closed 2023-02-28 by hughsie

dcd-arnold, 2023-02-28:
> Look what I spy! <requires><firmware compare="ge" version="1.5.0"/><client/></requires> -- that looks like a LVFS website issue as the root cause. Debugging now.

@hughsie the error within the data pulled from LVFS still exists for me.

hughsie, 2023-02-28:
Right, agree -- I'm testing it locally and will deploy it onto the staging env first. I'll take a few hours

hughsie, 2023-02-28:
@dcd-arnold component 17216 was the only one affected on the production LVFS. The commit was https://gitlab.com/fwupd/lvfs-website/-/commit/65b7e1c77e42aa0774de09d37ded92843ae10859 - if you do "fwupdmgr refresh --force" the issue should be gone. You might need to do `sudo rm /var/lib/fwupd/metadata/lvfs/metadata.xml.xz` to allow the daemon to start.

dcd-arnold, 2023-03-01:
@hughsie I can confirm that it is working now as expected. Thank you very much.

hughsie, 2023-03-01:
Phew, thanks for all the help.