fwupd/firmware-dell #159: Update failure on reboot

Vallas873, 2023-02-11:


Vallas873, 2023-02-18:
Hello, to give an update to this (lol) I just update my firmware with an USB stick (as explained [here](https://www.dell.com/support/kbdoc/fr-fr/000131486/update-the-dell-bios-in-a-linux-or-ubuntu-environment)). It worked well and `fwupmgr` detect that my system is now up to date.
`fwupmgr` didn't work though, so I let this issue open for now.

gsly, 2023-07-10:
@Vallas873 I am seeing the same issue here on an OptiPlex 7040 USFF. It is stuck on 1.9.2 and I can't seem to get any tool to upgrade it including `fwupmgr`, the USB boot method as well as the Flash BIOS Boot Menu.

They show all the correct BIOS update information (USB boot showing the most including ME version) but they all just reboot back into the same 1.9.2 BIOS version and I have never seen the "flashing..." screen I've seen researching this issue.

Ideas?

gsly, 2023-07-12:
I have discovered that my OptiPlex 7040 has the Intel Management Engine (ME) disabled from Dell as per the original configuration (I'm not the 1st owner). This explains why I see no AMT/ME options in the BIOS and likely why the BIOS update never starts.