fwupd/firmware-dell #160: Suspicious update offered for Dell XPS 9320

heftig, 2023-02-18:
I'm being offered a suspicious update for the Dell XPS 9320 system firmware which is not available via the support website (still at 1.11.0).

The description suggests it is for yet-to-be-released hardware and should not apply to my laptop, which has a 12th gen (Alder Lake), not a 13th gen (Raptor Lake) processor.

```
> fwupdmgr get-updates                   
Devices with no available firmware updates: 
 • VEN 04F3:00 04F3:31D1
 • 0000:00:1f.5
 • PC801 NVMe SK hynix 1TB
 • TPM 2.0
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB4 Retimer
Dell Inc. XPS 9320
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    1.11.0
  │   Minimum Version:    1.11.0
  │   Vendor:             Dell (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUIDs:              8b8c05a5-31bc-5fa4-9624-24b13f8f6043
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 9320:
        New version:      2.0.0
        Remote ID:        lvfs
        Release ID:       19486
        Summary:          Firmware for the Dell XPS 9320
        License:          Proprietary
        Size:             23.7 MB
        Created:          2023-02-06
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Initial Release for 13th generation Intel processors.
```

**Expected behavior**

Offered firmware updates match the ones available via Dell's support website.

**fwupd version information**

```
> fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.1.12-zen1-1-zen
```

Installed via Pacman on Arch Linux.
[... 191 lines cut ...]
│   
├─USB4 host controller:
│     Device ID:          bd45c8598c8ed529762ec4740ed0168dd8516324
│     Summary:            Unmatched performance for high-speed I/O
│     Vendor:             INTEL (THUNDERBOLT:0x8087)
│     GUIDs:              5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│                         d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Signed Payload
│   
└─USB4 host controller:
      Device ID:          98e99efb23380994929ff93cbd82229b9a395611
      Summary:            Unmatched performance for high-speed I/O
      Vendor:             INTEL (THUNDERBOLT:0x8087)
      GUIDs:              5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
                          d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
                          e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
```

</details>

hughsie, 2023-02-19:
@CragW can you have a look as a priority please.

CragW, 2023-02-20:
The release has just been corrected, @heftig please check at your convenience.

hughsie, 2023-02-20:
Thanks @CragW -- much appreciated.

closed 2023-02-20 by hughsie

heftig, 2023-02-20:
The suspicious update has disappeared and fwupd is back to reporting my firmware as up-to-date. Many thanks.

tcquinn, 2023-02-21:
@CragW I think this bad firmware update actually got installed on my system (Ubuntu 22.04 on Dell XPS 9320). On Friday, as part of the normal Ubuntu updates, I was notified that there was a BIOS update. The system updated BIOS automatically, and then I started having all kinds of issues (video glitches, camera not working, etc.). If I go into the boot menu (F12) now, my system reports a BIOS version of 2.0.0 (which shouldn't exist, right?). Dell tech support is not being helpful. Any advice?

hughsie, 2023-02-21:
@tcquinn can you `fwupdmgr downgrade` to the previous version? If not, @CragW needs to help I'm afriad.

heftig, 2023-02-21:
You can download [the current firmware (an .exe file)](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=gxfp1&oscode=w2021&productcode=xps-13-9320-laptop&lwp=rt) and copy it to the EFI System Partition, probably `/boot/efi/`. Then select the file with the BIOS Update utility you can access via F12 at boot.

tcquinn, 2023-02-21:
@hughsie @heftig Thanks for the suggestions. There was a whole directory tree of existing stuff under `/boot/efi` and I didn't want to screw any of that up, so I downloaded the current firmware (1.11.0) to a flash drive and tried to flash from that. I got a message that says "Firmware Update is not permitted. The BIOS flash is not compatible with the current system." (image of screen [here](https://www.dropbox.com/s/5z5nl4oi2ygq8ta/bios_update_not_permitted.jpg?dl=0) if helpful)

heftig, 2023-02-21:
Strange that it doesn't seem to be able to parse the file at all, reporting no system or revision for it. Was it damaged? It should have the sha256sum `c676ed424eb2f0c57bb82734da2096a79410fa662beaaa37664a5d8e7d257665`.

tcquinn, 2023-02-21:
@heftig Yes. `sha256sum XPS_9320_1_11_0.exe` yields `c676ed424eb2f0c57bb82734da2096a79410fa662beaaa37664a5d8e7d257665  XPS_9320_1_11_0.exe`

tcquinn, 2023-02-21:
I suppose it's possible that there's some problem with booting from a USB flash drive, but it found the flash drive and I was able to navigate to the file and select it.

heftig, 2023-02-21:
It's possible to disable BIOS downgrading in the setup. Maybe check that setting.

Putting a file into /boot/efi/ should be harmless as long as the partition has enough space left. The upgrade/downgrade itself will write some recovery files about 38M big.

Actually, try selecting the file `/EFI/Dell/bios/recovery/BIOS_PRE.rcv`, if it exists. That should be the previous firmware.

CragW, 2023-02-22:
I heard that 2.0.0 has the same components as 1.11.0, just the version increasing for a new branch. It is unlikely a fwupd issue here.

tcquinn, 2023-02-22:
@CragW Interesting. The problems are at a pretty basic hardware IO level (bands of random pixels across the screen when I move the mouse, camera stopped working) and they started right after this BIOS update installed. It's a strange coincidence. I'll try to dig further into the logs.

tcquinn, 2023-02-22:
@CragW In case you have two more seconds, here is the `apt` log from the moment when I think everything went haywire, in case anything catches your eye. Note the Dell hardware support packages updating.
```
Start-Date: 2023-02-18  09:35:34
Commandline: aptdaemon role='role-commit-packages' sender=':1.162'
Install: linux-modules-ipu6-5.19.0-32-generic:amd64 (5.19.0-32.33~22.04.1, automatic), linux-modules-ipu6-generic-hwe-22.04:amd64 (5.19.0.32.33~22.04.9, automatic), linux-modules-ivsc-5.19.0-32-generic:amd64 (5.19.0-32.33~22.04.1, automatic), linux-modules-ivsc-generic-hwe-22.04:amd64 (5.19.0.32.33~22.04.9, automatic)
Upgrade: containerd.io:amd64 (1.6.16-1, 1.6.18-1), language-pack-en-base:amd64 (1:22.04+20220721, 1:22.04+20230209), gnome-initial-setup:amd64 (42.0.1-1ubuntu2.2, 42.0.1-1ubuntu2.3), google-chrome-stable:amd64 (110.0.5481.77-1, 110.0.5481.100-1), firmware-sof-signed:amd64 (2.0-1ubuntu4, 2.0-1ubuntu4.1), alsa-ucm-conf:amd64 (1.2.6.3-1ubuntu1.3, 1.2.6.3-1ubuntu1.4), ubuntu-advantage-desktop-daemon:amd64 (1.9~22.04.1, 1.10~22.04.1), language-pack-en:amd64 (1:22.04+20220721, 1:22.04+20230209), language-pack-gnome-en-base:amd64 (1:22.04+20220721, 1:22.04+20230209), base-files:amd64 (12ubuntu4.2, 12ubuntu4.3), oem-somerville-tentacool-meta:amd64 (22.04ubuntu5, 22.04ubuntu6), language-pack-gnome-en:amd64 (1:22.04+20220721, 1:22.04+20230209), ubuntu-advantage-tools:amd64 (27.12~22.04.1, 27.13.5~22.04.1), oem-somerville-factory-tentacool-meta:amd64 (22.04ubuntu5, 22.04ubuntu6)
End-Date: 2023-02-18  09:35:59
```

MaikHerrmann, 2023-02-25:
@tcquinn The same thing happened on my Dell XPS 9320. Firmware got flashed through automatic update. I am using Fedora instead, but the issues are the same. The camera stopped working after the update to 2.0.0 and random horizontal pixel lines appear occasionally. Downgrading to 1.11.0 did not fix my camera issue. Is there a solution for that?

tcquinn, 2023-02-25:
@MaikHerrmann At least for me, the video artifacts turned out to be related not to the erroneous firmware update (confusingly), but rather to the (near-simultaneous) push of kernel version 5.19. There is a temporary fix [here](https://askubuntu.com/questions/1455653/horizontal-colored-lines-appearing-on-display-ubuntu-22-04) which might help even though you're on Fedora.

Zwyx, 2023-03-29:
Hey all, I have updated my firmware to `2.0.0`. **I don't have any issues***, should I downgrade my firmware anyway?

Basically, my question is: will Dell publish new versions `1.x.x` in the future? Because if `1.12.0` is published for instance, then I guess my system will ignore it.

---

\* apart from the video artifacts, but that's been proven unrelated.

sunnycrown83, 2023-03-29:
> Hey all, I have updated my firmware to `2.0.0`. **I don't have any issues***, should I downgrade my firmware anyway?
>
> Basically, my question is: will Dell publish new versions `1.x.x` in the future? [...]

Not necessary, there is no longer 1.x.x version for XPS 9320 in the future, you could stay with 2.0.0 and it's the latest version by now.