fwupd/firmware-dell #164: Optiplex 3080 "search" showing wrong latest version - remove extraneous leading zeros?

pjwelsh, 2023-02-28:
LVFS version  search page is showing "1.3.12" the latest:
https://fwupd.org/lvfs/search?value=optiplex+3080

However, clicking the page link *kinda* has the correct version "00.2.17.1" showing. The issue may be in the odd leading zeros? They appear to be extraneous and may be causing the "issue".
The official page does not have the leading zeros:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=t1yjp&oscode=biosa&productcode=optiplex-3080-desktop

CragW, 2023-03-07:
@hughsie is this a lvfs-website issue?

superm1, 2023-03-07:
No nothing to add. I agree drop inf parsing.

hughsie, 2023-03-07:
Okay, done. https://fwupd.org/lvfs/devices/com.dell.uefi10d19ee4.firmware now shows no leading zeros.

pjwelsh, 2023-03-07:
Will the "search" part update later? Currently it shows a 1.3.12 version as newest now:
https://fwupd.org/lvfs/search?value=optiplex+3080
Should be the "2.17.1" version.

CragW, 2023-03-08:
@hughsie Agree with you for the change to `inf` parsing, thanks.

With regarding to the `priority` attribute for a composite device.. the dock BKC turned out is a rich mixture of 3 different docks and each packages has its own version. Perhaps pick one of them as the main representative rather a random subcomponent for now, and see how it goes with the clarity.

pjwelsh, 2023-03-08:
Please re-open the ticket, the fundamental issue is still apparent - the wrong version is showing as latest when searched:
https://fwupd.org/lvfs/search?value=optiplex+3080
Should show the "2.17.1" version and not the 1.3.12 version shown.

hughsie, 2023-03-10:
Fixed in the latest LVFS deploy. Thanks all!

closed 2023-03-10 by hughsie