fwupd/firmware-dell #166: XPS 15 9550 wants to update precision 5510 bios?

Korothi, 2023-03-16:
i have dell xps 15 9550 using 1.14.0 bios and it just asked me to update precision 5510 to 1.23.0

hughsie, 2023-03-16:
@CragW is this the same BIOS?

Korothi, 2023-03-16:
[image]
some screenshot

CragW, 2023-03-29:
It seems they've fixed this issue in the metainfo, @Korothi would you want to try again?

dabrowne, 2023-04-03:
Can someone please help explain what's gone wrong here? I saw this update in the Ubuntu software updater UI and stupidly clicked install without noticing that it had the incorrect name. Surprisingly, the firmware install completed successfully but my system performance has been miserable since.

I've since reverted back to the most recent BIOS version 1.14.0 for the XPS 9550, however the Intel ME installation failed during the downgrade (I guess it's not possible to downgrade this) so I'm stuck with a potentially incompatible Intel Management Engine package for my system which I suspect is causing my performance issues. Is there any guidance on how this can be resolved?

hughsie, 2023-04-03:
@CragW seems like this isn't indeed fixed. Can you investigate please.

CragW, 2023-04-07:
Intel ME cannot get downgraded, not fwupd specific problem. @dabrowne would you please raise a ticket to support at dell.com and elaborate the performance issue for further analysis?

davidhall-enactor, 2023-06-13:
> @CragW seems like this isn't indeed fixed. Can you investigate please.

Just to confirm, I have the same problem and it doesn't appear to be fixed yet.

```
$ sudo fwupdmgr get-updates
Devices with no available firmware updates: 
 • BCM920703 Bluetooth 4.1
 • Sabrent Rocket Q
 • UEFI dbx
XPS 15 9550
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    1.14.0
  │   Minimum Version:    1.14.0
  │   Vendor:             Dell (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUIDs:              124c207d-5db8-4d95-bd31-34fd971b34f9
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  ├─Precision 5510:
  │     New version:      1.22.0
  │     Remote ID:        lvfs
  │     Release ID:       14400
  │     Summary:          Firmware for the Dell XPrecision 5510
  │     License:          Proprietary
  │     Size:             13.4 MB
  │     Created:          2022-06-06
  │     Urgency:          High
  │     Vendor:           Dell
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     This stable release fixes the following issues:
  │     
  │     • Improved the system stability.
  │   
  ├─Precision 5510:
  │     New version:      1.21.0
  │     Remote ID:        lvfs
  │     Release ID:       13391
  │     Summary:          Firmware for the Dell XPrecision 5510
  │     License:          Proprietary
  │     Size:             13.4 MB
  │     Created:          2022-03-23
  │     Urgency:          Critical
  │     Vendor:           Dell
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     This stable release fixes the following issues:
  │     
  │     • Firmware updates to address security vulnerabilities.
  │   
  └─Precision 5510:
        New version:      1.20.0
        Remote ID:        lvfs
        Release ID:       11553
        Summary:          Firmware for the Dell XPrecision 5510
        License:          Proprietary
        Size:             13.4 MB
        Created:          2021-12-08
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Firmware updates to address security vulnerabilities.
        • Firmware updates to address the Intel Security Advisory.
```

dabrowne, 2023-06-21:
This has come back for me too.. This morning I got a notification from the Ubuntu Software app that a "Critical Update" is ready to install:

[image]

CragW, 2023-06-26:
The update for Precision 5510 1.21.0 should have been removed from the catalog.

dabrowne, 2023-06-26:
> The update for Precision 5510 1.21.0 should have been removed from the catalog.

Thanks @CragW, this appears to be resolved again - I'm no longer seeing 1.21.0 as an available update.

davidhall-enactor, 2023-06-26:
Thanks @CragW the Precision 5510 firmware is not being displayed now. But I'm getting an update notification for a v1.15.0 BIOS which I'm not convinced is correct, as v1.14.0 that I already have is the latest listed on [Dell's website](https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=90khw&oscode=wt64a&productcode=xps-15-9550-laptop) for the XPS 15 9550. Can anyone else reproduce this?

```
$ sudo fwupdmgr get-updates
[sudo] password for david: 
Devices with no available firmware updates: 
 • BCM920703 Bluetooth 4.1
 • Sabrent Rocket Q
 • UEFI dbx
XPS 15 9550
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    1.14.0
  │   Minimum Version:    1.14.0
  │   Vendor:             Dell (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUIDs:              124c207d-5db8-4d95-bd31-34fd971b34f9
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 15 9550/Precision 5510:
        New version:      1.15.0
        Remote ID:        lvfs
        Release ID:       6405
        Summary:          Firmware for the Dell XPS 15 9550/Precision 5510
        License:          Proprietary
        Size:             9.9 MB
        Created:          2020-07-23
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Firmware updates to address the Intel Technical Advisory.
      
```

DPaulW, 2023-07-25:
Currently I have the same issue, Gnome Software is showing 1.15.0 as available for Dell XPS 9550 which it isn't. Any way to completely disentangle XPS 15 9550 and Precision 5510?
[image]

nahall, 2023-08-20:
Same here, KDE Plasma Discover is showing an update to the 1.15.0 BIOS for my XPS 9550 but [Dell's website shows 1.14.0 as being the latest](https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=90khw&oscode=wt64a&productcode=xps-15-9550-laptop)

It is pulling up the BIOS from the Precision 5510 it seems.

CragW, 2023-08-21:
This is what I've heard -> Both BIOS versions (1.14.0 and 1.15.0) are derived from the same source code. Just the update cycles differ due to variations in product lifecycles.

DPaulW, 2023-08-21:
The description in fwupdmgr is exactly that for 5510 ( Firmware updates to address the Intel Technical Advisory. Issues:           CVE-2020-8758  INTEL-TA-00404) I can't find this being fixed for xps 15 9550 anywhere so please remove this as an option for xps 15 9550 to avoid issues faced by @dabrowne

MartFab, 2023-08-30:
Sorry if it is a bit bit tiny of-topic but i just noticed, that my XPS 9550 bios has been updated to 5510 1.23.0 bios. Because I'm a causal user, i haven't notice major drawback for now but what impacts can i expect from it ? And if there is, how can we downgrade as Intel ME doesn't seems to support downgrade ?
Thanks!

davidhall-enactor, 2023-08-31:
I think the current status of this ticket is;

1. The original issue is that `fwupd` was showing system updates for the Precision 5510 on the XPS 9550. If a user updated (as @dabrowne did) there may have been a performance impact as a result. This issue is now resolved.

2. However, now `fwupd` is showing a v1.15.0 system update for the "XPS 15 9550/Precision 5510" on the XPS 9550, which isn't available on Dell's website. I don't know whether there are any known issues updating to this version and am not too keen to find out! @CragW says "Both BIOS versions (1.14.0 and 1.15.0) are derived from the same source code" but I'm not sure this resolves the matter. The question is should `fwupd` be listing the 1.15.0 system update for the XPS 9550 or not, and I don't think that's been definitively answered.

DPaulW, 2023-09-01:
Accurate summary. I would vote :- no, 1.15.0 should definitely not be listed

CragW, 2023-12-07:
> I've also just hit this, which raised a red flag because I hadn't seen it on Dell's website.

They have set 1.15.0 as a reliable release for Linux.

kevlawz, 2024-10-23:
I know this is an old issue, but it still prompts to install 1.15 on XPS 9550 and that causes issues as outlined below. 

I can confirm that upgrading Dell XPS 9550 to this 1.15 bios breaks my Windows 11 performance. The CPU gets set to a very slow max clock speed (1.6 GHz?). 

Some notes (Dual boot with Win 11):

1. Downgrading from 1.15 to 1.14 does not resolve the Win 11 CPU speed issue. I suspect that's because of IME refusing to downgrade. 

2. If I reinstall the Intel Thermal Management driver on Windows, the CPU speed stepping works as normal. Upon reboot, the slow max CPU speed returns. This is the only way to have the laptop CPU power management function correctly. 

3. I had success using 3rd party CPU speed utilities to manually increase the multiplier to lock in max CPU speed. Only practical when plugged in. 

4. I do not recall any CPU speed issues when booting into Linux with 1.15. I use the downgrade to 1.14 / IME mismatch, and it seems to be OK for Linux as far as I can tell. Automatic CPU speed adjustments seem to be working.