fwupd/firmware-dell #170: XPS 15 9560 - system firmware 1.32.0 - checksum invalid

flipflop97, 2023-05-08:
When trying to update the system firmware of my XPS 15 9560 from 1.31.0 to 1.32.0, I get an error that the checksum is invalid:

```
checksum invalid, expected 6f8461008c222b1e888f8a097f963ca1aeacd4371a8f6a2df7ad429942381481 got fd53933b96c117f21370821e7fc89b30d84176c48fbc616ac533249c2c40d45c
```

The file it is downloading is https://fwupd.org/downloads/fd53933b96c117f21370821e7fc89b30d84176c48fbc616ac533249c2c40d45c-Dell_Precision_5520_System_BIOS_Ver.1.32.0.cab

```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.14
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.3.1-arch1-1
runtime   org.freedesktop.fwupd         1.8.14
```

hughsie, 2023-05-08:
Mea culpa: https://github.com/fwupd/fwupd/issues/5795 -- can you fwupdmgr refresh --force please

flipflop97, 2023-05-08:
The invalid checksum error is now gone, but it now says `No supported devices found`, right after extracting. It doesn't seem to install anything.

hughsie, 2023-05-08:
Can you get a daemon log of the failure please? e.g. do `sudo /usr/libexec/fwupd/fwupd --verbose --verbose` in one terminal, and then `fwupdmgr update` in the other.

flipflop97, 2023-05-08:
I don't have a `libexec` folder, but I assume the one in the normal `lib` folder is the same:

```
$ sudo /usr/lib/fwupd/fwupd --verbose --verbose

...
10:40:54.725 FuMain               Called SetHints()
10:40:54.725 FuMain               got hint locale=nl_NL.UTF-8
10:40:54.725 FuMain               Called GetPlugins()
10:40:54.727 FuMain               Called SetFeatureFlags(511)
10:40:54.727 FuMain               Called GetDevices()
10:40:54.729 FuMain               Called GetUpgrades(a45df35ac0e948ee180fe216a5f703f32dda163f)
10:40:54.731 FuRelease            remote lvfs has kind=download and so marking as trusted
10:40:54.731 FuRelease            remote lvfs has kind=download and so marking as trusted
10:40:54.731 FuRelease            remote lvfs has kind=download and so marking as trusted
10:40:54.732 FuEngine             failed to set release for component: Specified firmware is older than the minimum required version '1.31.0 < 1.31.0'
10:40:54.734 FuEngine             230c8b18-8d9b-53ec-838b-6cfc0383493a was not found: no results for XPath query 'components/component['type',attr(),'firmware',eq()]/provides/firmware['type',attr(),'flashed',eq()text(),?0,eq()]/../..'
10:40:54.734 FuEngine             ignoring 1.31.0 == 1.31.0
10:40:56.223 FuMain               Called GetDevices()
10:40:56.228 FuMain               Called GetRemotes()
10:40:59.125 FuMain               Called Install(a45df35ac0e948ee180fe216a5f703f32dda163f,14)
10:40:59.125 FuMain               got option reason
10:40:59.130 FuDeviceList         no replug or re-enumerate required
10:40:59.131 FuMain               Emitting PropertyChanged('Status'='decompressing')
10:40:59.234 Jcat                 Jcat engine kind 'sha1' not allowed
10:40:59.235 Jcat                 Jcat engine kind 'sha1' not allowed
10:40:59.235 Jcat                 got 1 PKCS7 signatures
10:40:59.235 Jcat                 verified: JcatResult:
  Timestamp:             2023-05-08T02:57:03Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

10:40:59.235 Jcat                 using gpgme v1.20.0
10:40:59.252 Jcat                 Using engine at /var/lib/fwupd/gnupg
10:40:59.256 Jcat                 ignoring LVFS-CA.pem as not GPG public key
10:40:59.256 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware
10:40:59.261 Jcat                 importing key F2F0325648E1AE4956710198C6361787A0A849E1 [0] Succes
10:40:59.261 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
10:40:59.266 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Succes
10:40:59.266 Jcat                 ignoring LVFS-CA.pem as not GPG public key
10:40:59.266 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
10:40:59.271 Jcat                 importing key CAE0420259890141E701156E0A6DCEF6B46F3DBF [0] Succes
10:40:59.271 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
10:40:59.275 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Succes
10:40:59.280 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
10:40:59.280 Jcat                 verified: JcatResult:
  Timestamp:             2023-05-08T02:57:03Z
  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
  JcatGpgEngine:
    Kind:                gpg
    VerifyKind:          signature

10:40:59.280 FuCabinet            verified metadata firmware.metainfo.xml: 3
10:40:59.280 FuCabinet            processing file: firmware.metainfo.xml
10:40:59.281 FuCabinet            processing release: 0x12000
10:40:59.345 Jcat                 Jcat engine kind 'sha1' not allowed
10:40:59.411 Jcat                 Jcat engine kind 'sha1' not allowed
10:40:59.411 Jcat                 got 1 PKCS7 signatures
10:40:59.445 Jcat                 verified: JcatResult:
  Timestamp:             2023-05-08T02:57:03Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

10:40:59.490 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
10:40:59.490 Jcat                 verified: JcatResult:
  Timestamp:             2023-05-08T02:57:03Z
  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
  JcatGpgEngine:
    Kind:                gpg
    VerifyKind:          signature

10:40:59.490 FuCabinet            verified payload firmware.bin: 3
10:40:59.557 FuMain               Emitting PropertyChanged('Status'='idle')
```

```
$ fwupdmgr update

Devices with no available firmware updates: 
 • PC300 NVMe SK hynix 512GB
 • TPM 2.0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.31.0 to 1.32.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisory.                                                           ║
║                                                                              ║
║ XPS 15 9560 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Bezig met uitpakken...   [***************************************]
No supported devices found
```

hughsie, 2023-05-08:
Ohh, sorry -- i need the daemon log after the `fwupdmgr update` has run in the other terminal.

flipflop97, 2023-05-08:
That's what I included, I removed all logs before running `fwupdmgr update`. No further logs were generated after the end of that block.

hughsie, 2023-05-08:
Maybe this is a dupe of https://github.com/fwupd/fwupd/issues/5794 perhaps?

flipflop97, 2023-05-08:
Looks like it! Since the original issue is solved I'll close this, thanks for the quick response/fix!

closed 2023-05-08 by flipflop97