fwupd/firmware-dell #172: xps 17 9720 finding no updates

droidben, 2023-07-10:
**Describe the bug**
I have a Dell xps17 9720 running Ubuntu 22.04.2 with uefi and secureboot enabled. UEFI capsule updates are enabled in the BIOS. despite this, fwupd failes to update my firmwares. I have tried manually setting the `OverrideESPMountPoint` to `/boot/efi`, which didn't help. Yes, udisk is installed. My boot-disk is ZFS.

**Steps to Reproduce**
```
$ sudo fwupd.fwupdmgr refresh --force 
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
$ sudo fwupd.fwupdmgr get-updates 
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • VEN 04F3:00 04F3:311C
 • Fingerprint Sensor
 • Integrated Webcam HD
 • SSD 980 PRO 1TB
 • SSD 980 PRO 2TB
 • System Firmware
 • TPS DMC Family
 • TPS6598X PD#0
 • TPS6598X PD#1
 • TS4
 • UEFI Device Firmware
 • UEFI dbx
 • USB3.0 Hub
 • USB4 Retimer
 • USB4 Retimer
No updatable devices
```

**Expected behavior**
It should find some updates, since I know, my device isn't up2date.

**fwupd version information**
I'm using the latest stable snap version of fwupd

```
$ sudo fwupd.fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    5.19.0-46-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. XPS 17 9720
│
├─VEN 04F3:00 04F3:311C:
│     Device ID:          cc9c1b47c3126a46379087db3bf18fa324f317ae
│     Summary:            Touchpad
│     Current version:    0x000c
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              46598f8d-5777-50c4-8da9-cda0cc8bbc26 ← HIDRAW\VEN_04F3&DEV_311C
│                         9b6fef1a-77aa-565c-9983-9da2dd2c932b ← HIDRAW\VEN_04F3&DEV_311C&REV_00
│                         d00322f0-ff1f-5458-9c64-8a771fb0b72e ← HIDRAW\VEN_04F3&DEV_311C&MOD_0000
│                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
[... 1105 lines cut ...]
├── initrd.img -> initrd.img-5.19.0-46-generic
├── initrd.img-5.19.0-45-generic
├── initrd.img-5.19.0-46-generic
├── initrd.img.old -> initrd.img-5.19.0-45-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.19.0-45-generic
├── System.map-5.19.0-46-generic
├── vmlinuz -> vmlinuz-5.19.0-46-generic
├── vmlinuz-5.19.0-45-generic
├── vmlinuz-5.19.0-46-generic
└── vmlinuz.old -> vmlinuz-5.19.0-45-generic

60 directories, 743 files
```

**Additional questions**

- Operating system and version: Ubuntu 22.04.2
- Have you tried rebooting? yes
- Is this a regression?
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2023-07-10:
@CragW I think a HWID might be missing somewhere.

superm1, 2023-07-10:
> Yes, there is a Dell Precision variant of this laptop I believe. Am I right by understanding that LVFS needs to add the XPS HWIDs to their site or can I change my config so it works?
> I found these links:
> https://fwupd.org/lvfs/devices/com.dell.uefi7d606caf.firmware
> https://fwupd.org/lvfs/devices/com.dell.uefiaaaaa926.firmware
> https://fwupd.org/lvfs/devices/com.dell.uefia888c891.firmware

That's right the HWIDs must be added for the update to work.

CragW, 2023-07-11:
@droidben although the system firmware is compatible with XPS 17 9720 model but Linux is beyond the [scope ](https://www.dell.com/support/home/en-us/product-support/product/xps-17-9720-laptop/drivers)of supporting OS'es.

droidben, 2023-07-11:
I understand now, I think. Dell can't be bothered to add some HWID to their LVFS site, that's the problem. Can we escalate this to them somehow or is it not worth it?

CragW, 2023-07-12:
I think that'd be challenging to ask for the release for an unsupported (untested) device.