fwupd/firmware-dell #173: Unable to update Dell U3421WE monitor from GNU/Linux

paulmenzel, 2023-07-24:
For the Dell U3421WE monitor, Dell offers [firmware update A00-00][1] from July 17th, 2023, only running on Microsoft Windows.

> Windows OS Firmware Update Package (Ver. M2B102) for U3423WE monitor.
> 
> IMPORTANT NOTE: The monitor firmware update will take approximately 25 minutes. Do not run Dell Display Manager software during firmware updating. Interruptions during the update process such as disconnecting or turning off the monitor or PC will cause damage to your monitor. Read the Firmware Update Instructions/Readme found in the Firmware package before updating your monitor. Check the following path for instructions: C:\Dell\firmware\U3423WE\Dell_U3423WE_FirmwareUpdate_Instruction_Windows_en_A00.pdf
> 
> Fehlerkorrekturen und Verbesserungen
> 
> 1. Enhance Dell Power Button Sync feature.
> 2. To fix monitor compatibility issues.

[1]: https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=vrhht&oscode=w2021&productcode=u3423we-monitor

CragW, 2023-07-25:
They have a separate release for this monitor at [dell com](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=96dnt&oscode=ubt20&productcode=dell-u3421we-monitor&lwp=rt)

paulmenzel, 2023-07-25:
Thank you for sharing the URL. Previously, I selected *Ubuntu Linux Desktop Edition* and nothing was shown.

The deb package contains a firmware file and a graphical installer.

    $ file usr/share/Dell/firmware/bin/monitor-firmware-update-utility
    usr/share/Dell/firmware/bin/monitor-firmware-update-utility: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=5e8f08519b7bfbd6e87c2a0a3b2d08bc8873c5dd, with debug_info, not stripped
    $ ls -l usr/share/Dell/firmware/U3421WE/M2B106/U3421WE_M2B106_20220512\(ISP\).UPG
    -rwxr--r-- 1 user user 2846421 12. Mai 2022  'usr/share/Dell/firmware/U3421WE/M2B106/U3421WE_M2B106_20220512(ISP).UPG'

Also, currently, non-Ubuntu users are left out.

So, the goal for monitor updates should also be to be integrated into fwupd/LVFS. (I am editing the title accordingly.)

PS: The [current page](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=96dnt&oscode=ubt20&productcode=dell-u3421we-monitor&lwp=rt) only shows the strings below as supported.

*   Ubuntu® 12.04
*   Ubuntu® 11.04 - ALL
*   Ubuntu® 11.10
*   Ubuntu® 10.04
*   Ubuntu® Desktop 16.04
*   Ubuntu® 16.04
*   Ubuntu® 16.04.3 LTS
*   Ubuntu® 18.04.3 LTS
*   Ubuntu® Desktop 8.04
*   Ubuntu® Desktop 8.10
*   Ubuntu® Desktop 7.04
*   Ubuntu® Server 8.10
*   Ubuntu® 18.04 LTS
*   Ubuntu 20.04 LTS
*   Ubuntu 22.04 LTS
*   Ubuntu Desktop Edition 7.10
*   Ubuntu® Desktop 9.10
*   Ubuntu® Desktop 9.04
*   Ubuntu® Moblin
*   Ubuntu® UMD
*   Ubuntu® Core 16
*   Ubuntu 10.10