fwupd/firmware-dell #174: Firmware requires itself - Dell OptiPlex 7760_7460 AIO - Version 1.26.0

mungomat, 2023-07-25:
The [specified version](https://fwupd.org/downloads/701f3625cf01e3396b1e17c4830933adeaab27a0341fc8b747a68daeaa5fec62-firmware_OptiPlex_7760_7460_AIO_1.26.0.cab) cannot be installed as it requires itself

```shell
# dmidecode -s system-product-name 
OptiPlex 7460 AIO

# dmidecode -s bios-version
1.25.0

# fwupdmgr install 701f3625cf01e3396b1e17c4830933adeaab27a0341fc8b747a68daeaa5fec62-firmware_OptiPlex_7760_7460_AIO_1.26.0.cab 
Decompressing…           [***************************************]
Not compatible with firmware version 1.25.0, requires >= 1.26.0
```

The errors seems to be in [firmware.metainfo.xml](https://fwupd.org/downloads/701f3625cf01e3396b1e17c4830933adeaab27a0341fc8b747a68daeaa5fec62-firmware_OptiPlex_7760_7460_AIO_1.26.0.cab):
```shell
[...]
<requires>
  <firmware compare="ge" version="1.26.0"/>
</requires>
[...]
```
expected:
```shell
[...]
<requires>
  <firmware compare="ge" version="1.10.0"/>
</requires>
[...]
```
Note: [Dell requires Version `1.10.0` for older versions](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=9y9k4) - click on **Important Information**: 
> If the current BIOS version is earlier than 1.6.4, ensure that you update the BIOS to version 1.10.0 before updating the BIOS to version 1.25.0.

---

Unfortunately we found further similar errors in **firmware.metainfo.xml** for **OptiPlex 7470** systems:
- [OptiPlex 7470 - Version 1.22.0](https://fwupd.org/downloads/975ed7109dcb3f10dc73c855343d7b0b2b539b79dc14ab3a1d1d6357401b9aaf-firmware_OptiPlex_7770_7470_AIO_1.22.0.cab) requires `1.22.0`, expected: `1.5.0`
- [OptiPlex 7470 - Version 1.23.2](https://fwupd.org/downloads/5e8ee672cf21e35c27953bcf7fc61f57383b511e4cab26cfd09d54689bea0214-firmware_OptiPlex_7770_7470_AIO_1.23.2.cab) requires nothing, expected: `1.5.0`

Note: [Dell requires Version `1.5.0` for older versions](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=m5xmt):
> If the current BIOS version is earlier than 1.0.2, ensure that you update the BIOS to version 1.5.0 before updating the BIOS to version 1.23.1.

---

Thank you for this useful program! 🙏

CragW, 2023-07-26:
They have just delivered a fix in the metadata, can you pl. try again in a couple hours?

mungomat, 2023-07-27:
- It works now for [OptiPlex 7460 - 1.26.0](https://fwupd.org/downloads/ff4c825826b7725849d906a5e416f793757d7212cea3770690c089331d3f51b3-firmware_OptiPlex_7760_7460_AIO_1.26.0.cab) and [OptiPlex 7470 - 1.23.2](https://fwupd.org/downloads/5e8ee672cf21e35c27953bcf7fc61f57383b511e4cab26cfd09d54689bea0214-firmware_OptiPlex_7770_7470_AIO_1.23.2.cab)
- Still not working for [OptiPlex 7470 - 1.22.0](https://fwupd.org/downloads/975ed7109dcb3f10dc73c855343d7b0b2b539b79dc14ab3a1d1d6357401b9aaf-firmware_OptiPlex_7770_7470_AIO_1.22.0.cab)
- Closing, since we are not affected anymore.
- Thank you very much

closed 2023-07-27 by mungomat