fwupd/firmware-dell #175: DOWNLOADING WRONG BIOS UPDATE FILE FOR Dell Latitude 7390 2-in-1

saitamasahil, 2023-11-22:
## Hardware Information:
- **Hardware Model:**                              Dell Inc. Latitude 7390 2-in-1
- **Memory:**                                      16.0 GiB
- **Processor:**                                   Intel® Core™ i7-8650U × 8
- **Graphics:**                                    Intel® UHD Graphics 620 (KBL GT2)

## Software Information:
- **Firmware Version:**                            1.32.0
- **OS Name:**                                     Fedora Linux 39 (Workstation Edition)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               45.1
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.5.12-300.fc39.x86_64

So I got an update via GNOME Software and it said Version 1.34.3 and the installed version was 1.31.0. I updated & rebooted. It failed. Same issue happened when i tried updating using terminal.

**It is downloading wrong file**
I checked Dell website and the latest version was 1.32.0 which was released on 20 November 2023. You can check latest version from [official website](https://www.dell.com/support/home/en-in/product-support/product/latitude-13-7390-2-in-1-laptop/drivers) I downloaded that & updated successfully using pen drive. Rebooted device & still it was showing an update for bios. Well i tried again & it failed. I checked [fwupd website](https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.firmware) & found out that latest version is 1.32.0 but in older version it is showing Version 1.34.3 as you can see in website i provided and the file name for v1.34.3 is Latitude_E5x70_Precision_3510_BIOS_1.34.3.cab. So, it is downloading file for another model which is precision 3510 and yes it's latest version of bios is 1.34.3 as you can see here on [official website](https://www.dell.com/support/home/en-in/product-support/product/precision-m3510-workstation/drivers). I feel like this was the reason it kept failing. It is still showing me 1.34.3 update even i updated to latest 1.32.0 update. 

> As you can see in the screenshots, it is showing wrong model number in GNOME Software and on fwupd website wrong files are uploaded.
[image]
[image]
[image]
[image]
[image]
[image]

hughsie, 2023-11-22:
@CragW missing HWID?

CragW, 2023-11-23:
> @CragW missing HWID?

Yes missed the HWID.  They're working on a fix.

CragW, 2023-11-27:
@saitamasahil can you refresh metadata and try again please?
- support.dell.com
  - https://www.dell.com/support/home/en-us/product-support/product/latitude-13-7390-2-in-1-laptop/drivers
- fwupd.org
  - https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.7390.firmware

closed 2023-11-27 by saitamasahil