fwupd/firmware-dell #180: [Missing Firmware] Dell Inspiron 5415/5515 and Vostro 5415/5515 System BIOS

mc-k1, 2024-02-09:
- latest BIOS is 1.20.0: https://dl.dell.com/FOLDER11029501M/1/Inspiron_Vostro_5415_5515_1.20.0.exe
- latest BIOS at LVFS is none
- system: Dell Vostro 5415

mark-chandler, 2024-06-19:
* latest BIOS is 1.23.1: https://dl.dell.com/FOLDER11606701M/1/Inspiron_Vostro_5415_5515_1.23.1.exe
* latest BIOS at LVFS is none
* system: Dell Vostro 5415

mc-k1, 2024-06-19:
* latest BIOS is 1.23.1: https://dl.dell.com/FOLDER11606701M/1/Inspiron_Vostro_5415_5515_1.23.1.exe
* latest BIOS at LVFS is none
* system: Dell Vostro 5415

chinyakao, 2024-06-19:
Hi @mc-k1 

We would like to kindly remind you that this platform is not supported by the Linux version at this time. Therefore, there is no BIOS on LVFS. If you are using Linux, we recommend exploring alternative solutions such as the BIOS from dell.com as you pasted. 
We appreciate your understanding. Thank you for your cooperation.

mc-k1, 2024-06-19:
@chinyakao - no worries. I'm not looking for updates from LVFS yet. I'm just maintaining this ticket as a reminder that this firmware has not been done yet. I used a USB drive to do the update using the latest BIOS that I downloaded. I'll try to used the firmware-updater script in the future.