fwupd/firmware-dell #181: requirement 'hardware' also needs org.freedesktop.fwupd >= 1.0.8

stephanebill, 2024-02-20:
Precision 3551 with BIOS 1.25, trying to update to 1.26 with fwupd 1.9.13 on recent Opensuse Tumbleweed.
I downloaded the firmware from https://fwupd.org/lvfs/devices/com.dell.uefid10f3c80.firmware and
ran fwupmgr install <file.cab> since the laptop cannot connect to the internet directly.
I get the error in the title.

hughsie, 2024-02-20:
Hmm, that firmware looks like it has the requirement (both in the web-UI/db and also the XML file inside the archive) -- I assume you're using https://fwupd.org/downloads/825b234d57d2d92a56e50b9a0206b0af9bb6a281dea8189eb3a878830301e88e-Signed_1152921505697209055.cab ?

stephanebill, 2024-02-20:
Yes, this is the file I am using.
I did more tests.  I have an update script for the BIOS updates that verifies the power is plugged in, downloads the correct update file for the laptop model and then starts 'fwupdmgr install -y file.cab'
Everything works fine when the script is started by root but fails when initiated by a regular user with sudo.