fwupd/firmware-dell #182: Dell 7480 AIO missing BIOS 1.23.0

dcd-arnold, 2024-02-28:
**Describe the question**

Hi there,

I have configured to install only pre-approved updates. I approved version 1.23.0 for my Dell 7480 AIO. However, this version is not found by `fwupdmgr update` anymore. I only get versions `1.28.0`, `1.27.0`, and `1.26.1`. The file is still available in the LVFS: https://fwupd.org/lvfs/devices/com.dell.uefia3ef064d.firmware

Is that by design or an error?

**fwupd version information**

```shell
fwupdmgr --version

runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-69-generic
compile   com.hughsie.libjcat           0.1.4
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4
```

hughsie, 2024-02-28:
> only pre-approved updates

With `fwupdmgr set-approved-firmware`?

dcd-arnold, 2024-02-28:
I have set `ApprovalRequired=true` in `/etc/fwupd/remotes.d/lvfs.conf` and `ApprovedFirmware=4a47d532bab2d916399c472973fe43512900e295` 

in the `daemon.conf`. It worked for a long time. But today I found a `Dell 7480 AIO` which would not update. I investigated with `fwupdmgr get-updates -vv` and found: 

```
14:03:45:0534 FuMain               current version is 1.18.0: 1.28.0=not-approved, 1.27.0=not-approved, 1.26.1=not-approved
```

And the version `1.23.0` which I approved is not listed anymore.

hughsie, 2024-02-28:
Does the /etc/fwupd/remotes.d/lvfs.conf have a .gz ending for the metadata or a .xz one? The former is only including the 3 most recent versions to work around a bug in very old versions of fwupd.

dcd-arnold, 2024-02-29:
It does in fact have a `.gz` ending. Changing this to `.xz`, dropping the cache `rm /var/cache/fwupd/*`, and running `fwupdmgr refresh --force` yields an error though:
```
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 48302273be56f64333a19db187728e30c9a73faa
```

How do I proceed from here?

dcd-arnold, 2024-02-29:
That did not work either:
```
# rm -rf /var/cache/fwupd/*
# rm -rf /var/lib/fwupd/metadata/*
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 48302273be56f64333a19db187728e30c9a73faa
```

I searched for similar issues here and found, that validation is made by data from a `.jcat` file. So I tried:
```
# wget https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
# wget https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
# jcat-tool info firmware.xml.xz.jcat | grep Data
      Data:              c6218c687732c17e5e83cbde236bce27de4341ae
      Data:              b8d84d4118aee51db5c0739995611c97210dc8fb4a3ba464b45c554cbce555fe
      Data:              -----BEGIN PGP SIGNATURE-----
      Data:              -----BEGIN PKCS7-----

root at bikecenter-C07DY63 in ~ on production
# jcat-tool info firmware.xml.gz.jcat | grep Data
      Data:              2843d09d88150342fad7a27577e4ed93dbf6b40c
      Data:              e6cf20b1ced2b0c9a65f8248dc73cd112d52974b5e28a240e7d5cebaf7ca6b26
      Data:              -----BEGIN PGP SIGNATURE-----
      Data:              -----BEGIN PKCS7-----
```

None of these files carry this checksum. I do not get where this checksum is coming from. What do I do now?

hughsie, 2024-03-08:
Ahh yes. You should be able to `fwupdmgr refresh --force` now and get a smaller metadata file. Does that help?

dcd-arnold, 2024-03-08:
Afraid not. `fwupdmgr refresh` still shows a different expected checksum for `.xz`-file than the `jcat`:
```
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 5c088b3f0a21f8deeab2c9f4fa96203e45f2a4f8

# jcat-tool info firmware.xml.xz.jcat | grep Data
      Data:              4e4a299e21f20b1db82274d53629b508cf222b25
      Data:              7afca878b502502725d958ebeb1970bf0cd6f25d3c83e9d4d816113e32c00c4d
```
I still have no clue why that is.

dcd-arnold, 2024-03-08:
With the `gz`-Version I get only the last *two* versions now:
```
(fwupdmgr:7816): FuMain-DEBUG: 11:37:41.571: current version is 1.18.0: 1.29.0=not-approved, 1.28.0=not-approved
```

I assume this is due to your hotfix. 
However, with this file size limit in place, I will never be able to retrieve all versions for a device and be forced to check new firmwares a lot faster than I would like.

Is it somehow possible to retrieve metadata by device? Even manually (and place it in the path for update)? (I can not believe I am the first one to have this issue)

hughsie, 2024-03-08:
1.7.9 is now restricted to 2 releases per component -- so that it works at all. Ubuntu needs to update the fwupd package, or at least backport the important fixes. The xz metadata isn't going to be understood by that old fwupd version either.

> Is it somehow possible to retrieve metadata by device

This isn't really something the LVFS provides I'm afraid.

> I can not believe I am the first one to have this issue

Most people are running much newer versions of fwupd.

dcd-arnold, 2024-03-11:
@hughsie thank you for all the support. We will implement our own download procedure and use `fwupd install` to install the local file. That seems the best way moving forward at this point. I am not sure how to proceed with this issue at this point. Feel free to close it.

closed 2024-03-11 by hughsie