fwupd/firmware-dell #184: XPS 15 9530 (2023)

Damienhaver, 2024-04-23:
- A link to the firmware update at Dell support
  - [link](https://www.dell.com/support/home/da-. dk/drivers/driversdetails?driverid=2MNG1)
- A link to the previous firmware upgrade at LVFS
  -  There are no updates for the firmware at LVFS
- System
  - Dell XPS 9530 (2023)

CragW, 2024-04-24:
This platform doesn't support Linux.

closed 2024-04-24 by CragW

Damienhaver, 2024-04-24:
Sad that this is the official stance of Dell, we can still update our systems firmware directly through the bios, by downloading the .exe, placing it in the esp and update it through the f12 menu, so why not just distribute the firmware through LVFS...

dell-client-linux, 2024-04-25:
Sorry for the inconvenience, Dell always pursues certify Ubuntu OS on more and more Dell’s products, you could look up the certified devices at: Certified laptops | Ubuntu<https://ubuntu.com/certified/laptops?>.
Unfortunately XPS 15 wasn’t in the list which means the its BIOS wasn’t qualified on Linux, that’s the major reason Dell doesn’t intend to push it to field users.

Internal Use - Confidential

[EXTERNAL EMAIL]