fwupd/firmware-dell #185: T5820 missing BIOS update 2.35

glennmorris, 2024-05-08:
Hi,

System: Dell Precision 5820 Tower.

Dell Bios update version 2.35.0, severity critical, released 8 April 2024:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=vntdt&oscode=wt64a&productcode=precision-5820-workstation

As of 8 May 2024, this version is not present on the LVFS site (which still has 2.34.0 as latest):
https://fwupd.org/lvfs/devices/com.dell.uefi9f1ecab8.firmware

Thanks.

CragW, 2024-05-10:
@glennmorris Can you grab the .exe file and install with `/usr/share/fwupd/install_dell_bios_exe.py`? The Linux distro version has been EOL'd.

glennmorris, 2024-05-10:
install_dell_bios_exe.py worked, thanks.
(I'm not sure what you mean by "The Linux distro version has been EOL'd".)

glennmorris, 2024-05-18:
Still no version 2.35 on LVFS, but I see version 2.36 has now appeared on dell.com and LVFS. :man_shrugging:

closed 2024-05-18 by glennmorris