fwupd/firmware-dell #186: "Updating from an EXE without a capsule" doesn't work

glennmorris, 2024-05-10:
Hi,

The README contains a section "Updating from an EXE without a capsule" describing how to use `contrib/firmware_packager/install_dell_bios_exe.py`
to install from a local exe file when the LVFS is not up to date.

When I tried to use this on AlmaLinux 9.4, it failed with:

```
org.freedesktop.fwupd property OnlyTrusted, current value is True
setting configuration key OnlyTrusted to false
Fwupd.Client.modify_config() takes exactly 4 arguments (5 given)

```

This recent [commit](https://github.com/fwupd/f) in fwupd self admittedly breaks the API, meaning install_dell_bios_exe.py from the repository is unlikely to work with any installed release of fwupd.

Reverting my local fwupd checkout to the previous revision allowed install_dell_bios_exe.py to work for me.

CragW, 2024-05-10:
The helper script from git source leans on the latest fwupd GObject to work along with new API.  If you have a `install_dell_bios_exe.py` script bundled in `fwupd` package that is carried by the Linux distro that should still work.

glennmorris, 2024-05-10:
Thank you, yes, using the bundled version `/usr/share/fwupd/install_dell_bios_exe.py` should avoid such compatibility issues.
Maybe the README in this repository could be updated to suggest that - at the moment, it just says to clone the fwupd repository and use the version from there.