fwupd/firmware-dell #188: Dell WD22TB4 upgrade issue - external display dropping / flickering randomly after the upgrade

tjanez, 2024-05-24:
**Describe the bug**

I've updated my Dell WD22TB4 dock's components firmware to their latest versions and after the upgrade, the external display connected through HDMI started dropping / flickering randomly, I would say more than once per minute.

Previously, I had no external monitor issues with WD22TB4.
And if I connect the external monitor to the Precision 5540's HDMI port directly, the dropping / flickering issues are gone.

Here are the versions of the Dell WD22TB4 dock's components before and after upgrade:

```shell
$ fwupdmgr get-history
Dell Inc. Precision 5540
│
├─Package level of Dell dock:
│ │   Device ID:          a55c2e5fd07e2a87a40bf7eda6c335da1a78e7e7
│ │   Previous version:   01.00.32.01
│ │   Update State:       Success
│ │   Update Error:       Pending activation
│ │   Last modified:      2024-05-21 06:43
│ │   GUID:               af2a3b39-98c4-59a9-9cbe-a6909a7d14a2
│ │   Device Flags:       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │                       • Unsigned Payload
│ │ 
│ └─(null) Update:
│       New version:      01.00.36.01
│       Remote ID:        lvfs
│       Description:      
│       The vendor did not supply any release notes.
│     
├─RTS5413 in Dell dock:
│ │   Device ID:          f2a41e8a079185a32ecbe129cead386b4dfe5a8d
│ │   Previous version:   01.21
│ │   Install Duration:   13 seconds
│ │   Update State:       Success
│ │   Last modified:      2024-05-21 06:43
│ │   GUID:               b27d25f1-019d-5718-b41a-02ddaefe5577
│ │   Device Flags:       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │                       • Signed Payload
│ │ 
│ └─(null) Update:
│       New version:      01.22
│       Remote ID:        lvfs
│       Description:      
│       The vendor did not supply any release notes.
│     
├─RTS5487 in Dell dock:
│ │   Device ID:          fd32487a5eec93ecf0aa9fddbe870795344b8ab6
│ │   Previous version:   01.49
│ │   Install Duration:   2 seconds
│ │   Update State:       Success
│ │   Last modified:      2024-05-21 06:43
│ │   GUID:               acfcd89b-105d-55b9-b85b-08bf8508f38c
│ │   Device Flags:       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │                       • Signed Payload
│ │ 
│ └─(null) Update:
│       New version:      01.57
│       Remote ID:        lvfs
[... 285 lines cut ...]
│     GUIDs:              d92f2014-425f-5023-b877-0ff468763d68 ← THUNDERBOLT\VEN_00D4&DEV_0906
│                         42598a87-0ee2-5d7e-b1d1-02a312041fd5 ← TBT-00d40906-native
│                         1d962ba4-9f85-5510-a85f-e4c94abf9b1d ← TBT-00d40906-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│                         • Signed Payload
│   

```

</details>

**Peripherals connected to the dock**
- Ethernet cable for networking
- HDMI for external monitor

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried unplugging the dock or any peripherals from your machine? Yes.
- Have you tried to power cycle the dock from the AC adapter? Yes.
- Is this a regression? Yes.

CragW, 2024-05-24:
@tjanez  Can you contact Dell support for further analysis on this isuse? This might be firmware issue instead fwupd's fault.

vinc17fr, 2024-05-29:
@tjanez You're using a 6.8 kernel, for which I and another user also have issues with external displays (no such issues with previous kernel versions):
* [Debian bug 1072063](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1072063) (I'm also using the WD22TB4 dock)
* https://github.com/pop-os/pop/issues/3261 (for this one, apparently without a dock)

Matthew1471, 2025-05-31:
> - Fixed the issue where the external monitor display is distorted when connected to the system through the Dell dock WD19 or WD22TB4.

https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=994jt&oscode=wt64a&productcode=wd22tb4-dock

vinc17fr, 2025-06-01:
I have version 01.00.47.01, and the issue still occurs.