fwupd/firmware-dell #195: Updates available for Dell OptiPlex 7450 All-In-One

mungomat, 2024-10-08:
- [Current version from Dell](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=xm6nd&oscode=wt64a&productcode=optiplex-7450-aio): `1.34.0`
- [Current version from LVFS](https://fwupd.org/lvfs/search?value=OptiPlex+7450): `1.30.0`
- System: **OptiPlex 7450 All-In-One**

hughsie, 2024-10-08:
What's the output for `fwupdmgr get-devices` please? We only need the section with the UEFI system bios, you can cut the other bits.

mungomat, 2024-10-09:
```
root at bikecenter-GF1GMV2 in ~ on internal
# fwupdmgr get-devices
OptiPlex 7460 AIO
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    73472
│ │   Minimum Version:    73472
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              ad1841ab-4452-4c34-bc1b-ed250cdaaf86
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
[...]
```

mungomat, 2024-10-09:
```
# fwupdmgr get-devices
OptiPlex 7450 AIO
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    73216
│ │   Minimum Version:    73216
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              3c20b9e1-eddd-4507-acd8-aa0d263f8cf6
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
[...]
```

CragW, 2024-10-11:
Ubuntu 16.04 lifecycle issue, same as #196.

closed 2024-10-22 by CragW

dcd-arnold, 2024-11-12:
@CragW does this mean that updates will not be provided to the LVFS going forward?