fwupd/firmware-dell #196: Updates available for Dell OptiPlex 7460 All-In-One

mungomat, 2024-10-08:
- [Current version from Dell](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=h6trt&oscode=wt64a&productcode=optiplex-7460-aio): `1.35.0`
- [Current version from LVFS](https://fwupd.org/lvfs/search?value=OptiPlex+7460): `1.31.0`
- System: **OptiPlex 7460 All-In-One**

Note: [Version 1.35.0 from OptiPlex 7460 AIO System Update](https://fwupd.org/lvfs/devices/com.dell.uefie8df61f8.firmware) does **not** work for us:
```
root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# dmidecode -s system-product-name
OptiPlex 7460 AIO

root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# dmidecode -s system-serial-number
GF1GMV2

root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.5.0-35-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# wget https://fwupd.org/downloads/19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b784e1bcc-firmware.cab
--2024-10-08 11:14:17--  https://fwupd.org/downloads/19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b784e1bcc-firmware.cab
Resolving fwupd.org (fwupd.org)... 2600:1f14:414:5602::4c9a, 2600:1f14:414:5601::b9bf, 54.149.25.170, ...
Connecting to fwupd.org (fwupd.org)|2600:1f14:414:5602::4c9a|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17317278 (17M) [application/vnd.ms-cab-compressed]
Saving to: ‘19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b784e1bcc-firmware.cab’

19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b7 100%[====================================================================================================================================>]  16.51M  5.71MB/s    in 2.9s    

2024-10-08 11:14:21 (5.71 MB/s) - ‘19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b784e1bcc-firmware.cab’ saved [17317278/17317278]

root at bikecenter-GF1GMV2 in /etc/fwupd on internal
# fwupdmgr --no-reboot-check install 19cea7db3b6559eb15ba1e28ca21451d2ab4fe10e1cd3726e647114b784e1bcc-firmware.cab
Decompressing…           [***************************************]
No supported devices found
```

hughsie, 2024-10-08:
What's the output for `fwupdmgr get-devices` please? We only need the section with the UEFI system bios, you can cut the other bits.

CragW, 2024-10-08:
The product with Ubuntu 16.04 has reached the end of support lifecycle. Perhaps you want to try the .exe install script.

mungomat, 2024-10-09:
> What's the output for fwupdmgr get-devices please? We only need the section with the UEFI system bios, you can cut the other bits.
```
root at bikecenter-GF1GMV2 in ~ on internal
# fwupdmgr get-devices
OptiPlex 7460 AIO
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    73472
│ │   Minimum Version:    73472
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              ad1841ab-4452-4c34-bc1b-ed250cdaaf86
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
[...]
```

> The product with Ubuntu 16.04 has reached the end of support lifecycle. Perhaps you want to try the .exe install script.

However, this machine is currently running Ubuntu 22.04.
Most of our devices have already been shipped, so we require a remote method to update the firmware.
`fwupdmgr` has been doing an excellent job for us — thank you for that.

CragW, 2024-10-11:
During the `.exe` install time a transient file `firmware.cab` will be generated under the temporary direcotry used for fwupd install, perhaps you want to save it before the system reboot?

See below  `/tmp/tmps7_iwu3m/firmware.cab` for your reference.

```
$ sudo ./install_dell_bios_exe.py ~/foo.exe
org.freedesktop.fwupd property OnlyTrusted, current value is True
setting configuration key OnlyTrusted to false
org.freedesktop.fwupd property OnlyTrusted, current value is False
Installing to System Firmware
Wrote capsule /tmp/tmps7_iwu3m/firmware.bin
GUID:      3a988588-49a8-4bea-8f42-a5b04b18efc4
HdrSz:     0x1000
Flags:     0x70000
PayloadSz: 0x3a41f98
Generated CAB file /tmp/tmps7_iwu3m/firmware.cab

org.freedesktop.fwupd property OnlyTrusted, current value is False
setting configuration key OnlyTrusted to true
org.freedesktop.fwupd property OnlyTrusted, current value is True
An update requires a reboot to complete
Restart now? (Y/N) n
Reboot your machine manually to finish the update.
```

closed 2024-10-22 by CragW