fwupd/firmware-dell #197: Missing Dell firmware update for Dell Latitude 9440

TommyTran732, 2024-10-10:
Latest version from Dell: [1.14.0](https://www.dell.com/support/home/en-us/product-support/product/latitude-14-9440-2-in-1-laptop/drivers)
Latest version from LVFS: [1.6.1](https://fwupd.org/lvfs/devices/com.dell.uefi412c9ac4.firmware)

Both the Latitude 9450 and 9430 gets regular updates, so I think Dell just forgot to maintain updates for the 9440.

TommyTran732, 2024-10-15:
@CragW Is there a reason why the generation before and after it have Linux support but not this one? Where can I find the list of devices that is expecting Linux support?

CragW, 2024-10-16:
> Where can I find the list of devices that is expecting Linux support?

This information should be somewhere at dell.com?

closed 2024-10-22 by CragW