fwupd/firmware-dell #199: FW update version 1.31.0 missing for Optiplex 7060

abhi-deshpande, 2024-10-21:
System - Optiplex 7060 (Tower, SFF, Micro)

Latest firmware version - 1.31.0
[Link](https://www.dell.com/support/home/en-in/drivers/driversdetails?driverid=fnjwn&oscode=wt64a&productcode=optiplex-7060-micro)

LVFS firmware version - 1.29.0
[Link](https://fwupd.org/lvfs/devices/com.dell.uefidf35f706.firmware)

It seems that no firmware updates for this device has been published on LVFS since a couple of quarters.

CragW, 2024-10-22:
Ubuntu 16.04 reached the end of lifecycle support, try `install_dell_bios_exe.py` instead.

closed 2024-10-22 by CragW

abhi-deshpande, 2024-10-22:
> Ubuntu 16.04 reached the end of lifecycle support, try `install_dell_bios_exe.py` instead.

Does that mean 7060 would not receive future FW updates via LVFS ?