fwupd/firmware-dell #200: [Question] Are TPM Firmware updates supported via LVFS ?

abhi-deshpande, 2024-10-21:
Are TPM Firmware updates supported by LVFS ? 

(Although these updates are not frequently published by Dell like BIOS ones.)

hughsie, 2024-10-21:
Yes, but I've never seen a vendor actually use the TPM-defined update protocol. Most TPM updates are performed using a UEFI UpdateCapsule or shipped as part of a composite system update in the case of Dell.

hughsie, 2024-10-21:
Which Dell product are you using? And what TPM version is installed and available?

abhi-deshpande, 2024-10-21:
Mine is an Optiplex 7060 micro with Nuvoton TPM 2.0. I remember I had upgraded TPM with TPM Firmware update utility from Dell Support before migration to Linux. Current version is the latest one 7.2.2.0. I asked this question because I am concerned with any future updates to it and how to do that on Linux.

fwupdmgr does my TPM as an updatable device.

hughsie, 2024-10-21:
@CragW can you help here? Thanks.

CragW, 2024-10-23:
TPM updates not available through LVFS at this time.

closed 2024-12-23 by CragW