fwupd/firmware-dell #201: dell firmware update corrupted bios dell xps 13 (9310) laptop

24fpsDaVinci, 2024-11-22:
this https://fwupd.org/lvfs/devices/com.dell.uefiecfae66b.firmware update that was pushed out on 11/15 to dell xps 9310 corrupted my bios. Not sure if its firmware issue or fwupd issue. It was fixed by installing the windows version of the same bios that was pushed out 3 days later by a shop (dont know how they did it because bios logs did not show any new firmware updates). Using firmware recovery and or downgrading did not work.

right after the initial fedora update, the laptop went into a boot loop and bios was no longer accessible.  the screen shows the dell logo and the diagnostic led flashes 3 orange, 6 white 

[image]
(https://www.dell.com/support/manuals/en-us/xps-13-9310-laptop/xps-13-9310-service-manual/system-diagnostic-lights?guid=guid-100544e8-6b7c-4da1-b749-7bcced1d8ca5&lang=en-us) 

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.20
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.1
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.0-65.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
as included in fedora rawhide

**Additional questions**

- Operating system and version:
 fedora rawhide
- Have you tried rebooting?
 yes
- Is this a regression?
 yes

CragW, 2024-11-22:
have you tried recovery?
https://www.dell.com/support/kbdoc/en-us/000132453/how-to-recover-the-bios-on-a-dell-computer-or-tablet

24fpsDaVinci, 2024-11-22:
yes recovery was tried. After recovery it will not boot pass the Dell logo. This screenshot was taken after the bios was fixed by a repair shop. Dell Recovery is always recorded in the bios logs. The repair shop used external eeprom flashing tools that was not recorded.

[image]

CragW, 2024-12-04:
Unlikely a fwupd issue, perhaps reach out to Dell support for further assistance in diagnosing this issue.

closed 2024-12-05 by CragW

24fpsDaVinci, 2024-12-05:
> Unlikely a fwupd issue, perhaps reach out to Dell support for further assistance in diagnosing this issue.

could this be caused by a bad bios release? and where would I report such a thing.

24fpsDaVinci, 2024-12-19:
looks like I was a victim of a bad bios release thats addressed in the 12.17.2024 release:
https://fwupd.org/lvfs/devices/com.dell.uefiecfae66b.firmware

```
 Version 3.27.1
2024-12-17 09:22:33
Fixed the issue where the BIOS Auto Recovery function always displays three Amber and six White diagnostic LEDs. This issue occurs after you enable the Intel Trusted Execution Technology (TXT) in the BIOS Setup.
```