fwupd/firmware-dell #203: Latest Optiplex 7070 BIOS version not finable via fwupd?

pjwelsh, 2025-01-06:
https://fwupd.org/lvfs/search?value=OptiPlex+7070
shows the latest 1.30.1 BIOS:
https://fwupd.org/lvfs/devices/com.dell.uefib7006d41.firmware

However, running fwupd does not see that update:

fwupdmgr --force --assume-yes refresh

fwupdmgr get-remotes 
Dell Inc. OptiPlex 7070
│
├─Vendor (Automatic):
│     Remote ID:          vendor-directory
│     Type:               directory
│     Keyring:            none
│     Enabled:            true
│     Priority:           1000
│     Filename:           /usr/share/fwupd/remotes.d/vendor/firmware
│     Filename Source:    /etc/fwupd/remotes.d/vendor-directory.conf
│   
├─Linux Vendor Firmware Service (testing):
│     Remote ID:          lvfs-testing
│     Type:               download
│     Keyring:            jcat
│     Enabled:            false
│     P2P Metadata:       true
│     P2P Firmware:       false
│     Refresh Interval:   24 hours
│     Priority:           1
│     Filename:           /var/lib/fwupd/metadata/lvfs-testing/metadata.xml.xz
│     Filename Signature: /var/lib/fwupd/metadata/lvfs-testing/metadata.xml.xz.jcat
│     Filename Source:    /etc/fwupd/remotes.d/lvfs-testing.conf
│     Metadata URI:       https://cdn.fwupd.org/downloads/firmware-testing.xml.xz
│     Metadata Signature: https://cdn.fwupd.org/downloads/firmware-testing.xml.xz.jcat
│     Report URI:         https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting:false
│   
└─Linux Vendor Firmware Service:
      Remote ID:          lvfs
      Type:               download
      Keyring:            jcat
      Enabled:            true
      P2P Metadata:       true
      P2P Firmware:       false
      Checksum:           3190f155d5ccf1de58f8fc28729ec48965210d89a8082c8aeabc3b40ac074088
      Age:                47 minutes
      Refresh Interval:   24 hours
      Filename:           /var/lib/fwupd/metadata/lvfs/metadata.xml.xz
      Filename Signature: /var/lib/fwupd/metadata/lvfs/metadata.xml.xz.jcat
      Filename Source:    /etc/fwupd/remotes.d/lvfs.conf
      Metadata URI:       https://cdn.fwupd.org/downloads/firmware.xml.xz
      Metadata Signature: https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
      Report URI:         https://fwupd.org/lvfs/firmware/report
      Automatic Reporting:false

fwupdmgr --assume-yes get-updates
Devices with no available firmware updates: 
 • HP SSD EX900 250GB
 • TPM
Devices with the latest available firmware version:
 • System Firmware
 • UEFI dbx
No updates available

fwupdmgr get-devices
Dell Inc. OptiPlex 7070
│
├─Core™ i5-9500 CPU @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543 ← CPUID\PRO_0&FAM_06&MOD_9E&STP_A
│     Device Flags:       • Internal device
│   
├─HP SSD EX900 250GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    S0614B0
│     Vendor:             Silicon Motion, Inc. (NVME:0x126F)
│     Serial Number:      HBSE29312300415
│     GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
│                         165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
│                         8a9e1a99-c66c-5fd9-bb56-e527c1c3acd6 ← HP SSD EX900 250GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1.29.0
│ │   Minimum Version:    1.29.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUID:               aecf8a53-cd41-4c16-8744-bd8e3725e28e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    7.2.1.0
│     Vendor:             Nuvoton Technology (TPM:NTC)
│     GUIDs:              fac1c8f3-73c8-5cd6-8330-07a3690b5140 ← TPM\VEN_NTC&DEV_0000
│                         8d94eb9a-78aa-513d-af50-319e3b9f5210 ← TPM\VEN_NTC&MOD_NPCT75x$rls
│                         e9ccc1dc-960a-5e09-afe9-e59a904b776d ← TPM\VEN_NTC&DEV_0000&VER_2.0
│                         f1c4013d-3f4c-5623-a40b-81ad96e23860 ← TPM\VEN_NTC&MOD_NPCT75x$rls&VER_2.0
│                         b98cbde6-72da-5f98-b56a-a8ea0ff3910e ← 092e-2.0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload

Capsule updates are allowed:
cat /sys/devices/virtual/firmware-attributes/dell-wmi-sysman/attributes/CapsuleFirmwareUpdate/current_value
Enabled

hughsie, 2025-01-07:
`com.dell.uefib7006d41.firmware` is _an_ OptiPlex 7070, but it's not _your_ OptiPlex 7070 -- I have no idea why OEMs insist on re-using the same model name for different models. You need the one with GUID  `aecf8a53-cd41-4c16-8744-bd8e3725e28e` -- for @CragW the link is https://fwupd.org/lvfs/firmware/99528 although there are quite a few other firmwares still sitting in the embargo remote too.

pjwelsh, 2025-01-26:
@CragW any comment for @hughsie regarding this firmware issue?

hughsie, 2025-03-10:
1.31.1 got automatically demoted due to a large number of failures on real world systems. It seems to deploy correctly, but the new version doesn't seem to report success reliably.

CragW, 2025-03-10:
>  at least one did not run the update.

Do you have a physical external display connected to the system?

> I wonder if the new version was earlier present but was withdrawn for some reason?

Re-promoted 1.31.1 aecf8a53 to stable, should be available in few hours.

> Is b7006d41 a real Dell machine? 

That's true.

imc0, 2025-03-11:
Updates have now been received.  Many thanks for the quick response.  A few success reports have been uploaded.

CragW, 2025-03-12:
Wonderful, growing successes! Should we close this issue?

pjwelsh, 2025-03-12:
OK with me at this point. Thank you for the help!
PJ

-- 

*Disclaimer*

The information contained in this communication from the 
sender is confidential. It is intended solely for use by the recipient and 
others authorized to receive it. If you are not the recipient, you are 
hereby notified that any disclosure, copying, distribution or taking action 
in relation of the contents of this information is strictly prohibited and 
may be unlawful.

This email has been scanned for viruses and malware, and 
may have been automatically archived. Cassens

closed 2025-03-13 by CragW