fwupd/firmware-dell #205: missing TPM and Thunderbolt firmware for Dell Precision 3551

brjsp, 2025-03-13:
<a href="https://dl.dell.com/FOLDER11323524M/1/DELLTPM_STHE4_1.769_64_v3.exe">TPM firmware</a> dated 2024-05-27. The .exe file can be run within Windows PE.

fwupd dump after update:
```
├─TPM:
│     Device ID:          1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
│     Current version:    1.769.0.0
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUIDs:              3680fbf1-593f-586f-91ac-c528b37e8373 ← TPM\VEN_STM&DEV_0000
│                         df479455-9790-58c9-b22a-bce5bc9eb149 ← TPM\VEN_STM&MOD_ST33TPHF2XSPI
│                         8d0b4adc-a42f-59eb-9df8-665923afa086 ← TPM\VEN_STM&DEV_0000&VER_2.0
│                         d565c809-9cf5-51c1-9ab7-d66b9d70ffbd ← TPM\VEN_STM&MOD_ST33TPHF2XSPI&VER_2.0
│                         0da67826-8c4a-5338-9165-192ac192eef9 ← 09c2-2.0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│                         • Can tag for emulation
│
```

<a href="https://dl.dell.com/FOLDER10502992M/2/Intel-Thunderbolt-Controller-Driver_TBB12_WIN_1.41.1379.0_A18_01.EXE">Thunderbolt firmware</a> dated 2025-03-07. The file does <b>not</b> run within WinPE, nor can be extracted using any tool i know.

fwupd dump <b>before</b> update, as i was unable to apply it (I have no windows install on that computer):
```
├─Thunderbolt host controller:
│     Device ID:          fa3d37c0f9c330671bca7568417affa9ffbbf19e
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    56.00
│     Vendor:             Dell (THUNDERBOLT:0x00D4, TBT:0x00D4)
│     GUIDs:              c4a71d83-8910-5ec9-99da-04987443e3f7 ← THUNDERBOLT\VEN_00D4&DEV_09C2
│                         8174f0f2-7f99-5960-addb-e6741ab9fd35 ← TBT-00d409c2-native
│                         95f442c5-a0e8-5271-b0ee-74c3532f79bf ← TBT-00d409c2-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│                         • Signed Payload
│                         • Can tag for emulation
│
```

CragW, 2025-03-17:
hi @brjsp , can you provide full device list please?

```
$ fwupdmgr get-devices --show-all-devices
```

CragW, 2025-03-17:
> [TPM firmware](https://dl.dell.com/FOLDER11323524M/1/DELLTPM_STHE4_1.769_64_v3.exe) dated 2024-05-27. The .exe file can be run within Windows PE.
> 

AFAIK no TPM updates for Linux.

> [Thunderbolt firmware](https://dl.dell.com/FOLDER10502992M/2/Intel-Thunderbolt-Controller-Driver_TBB12_WIN_1.41.1379.0_A18_01.EXE) dated 2025-03-07. The file does not run within WinPE (complains about missing driver — the driver can be installed from the .inf but fails to get loaded), nor can be extracted using any tool i know.
> 

This URL is thunderbolt driver.  You might be looking for this relesae? https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=tcrd4&oscode=wt64a&productcode=precision-15-3551-laptop

brjsp, 2025-03-17:
> This URL is thunderbolt driver. You might be looking for this relesae? 
Yes, sorry.