fwupd/firmware-dell #213: Alienware firmware update prevents Fedora kernels from booting

spotaws, 2025-04-16:
I have an Alienware M16 R2 system (can provide service tag on request), it is setup to dual-boot Windows 11 and Fedora 42, and this has been working fine until I booted into Windows 11 recently, where it updated the BIOS firmware from 1.9.0 to 1.10.0. After that change, Fedora kernels would not boot. Specifically, grub loads and presents the options, but none of the kernels will boot successfully. The same failure occurs if I attempt to boot a fresh Fedora image from a USB drive (I tried Fedora 39, 40, 41, 42, all without success). When I select a boot option from Grub, the screen goes black, except for a non-blinking cursor in the top left. After a few seconds, the keyboard lights go solid blue (from the rainbow pattern they were before) and USB devices connected lose power (I have an external keyboard/mouse). The system appears to be frozen.

In BIOS, Secure Boot is disabled and the RAID setting on storage is also disabled.

I can boot Ubuntu 24 and the latest Open SUSE live images via USB without issue, and I was able to mount my Fedora installation via chroot through this. I took the Open SUSE kernel and installed it into my Fedora environment, added it to grub, and that boots fine (and gets me back into my Fedora environment).

I tried to downgrade the BIOS firmware to 1.9.0, but it would not permit this.

TLDR: Something in the 1.10.0+ firmware is preventing Fedora kernels across the last several releases from booting on my Alienware M16 R2, where this previously worked.

hughsie, 2025-04-16:
@CragW can you check internally what was done in 1.10.0 please.

johnvanham, 2025-04-23:
Also having the exact same problem. Other linux distros all boot fine but not fedora after the BIOS update to 1.10.0 and there appears to be no way to downgrade - have tried a number of methods.

Other folks have same problem: https://discussion.fedoraproject.org/t/after-a-firmware-update-on-an-alienware-m16-r10-to-version-1-10-0-fedora-41-no-longer-boots/148158/1

CragW, 2025-04-23:
The update has been successfully verified on a machine-supported OS, and the same patch has been cross-checked on another system running Linux. It is unlikely that this is a fwupd issue that can be resolved here; Fedora's assistance may be required.

spotaws, 2025-04-29:
Happy to provide assistance here. I have a strong suspicion that something in Fedora's kernel sources or configuration is preventing it from working properly after the firmware update, but I have no idea where to start looking.

johnvanham, 2025-04-30:
> Happy to provide assistance here. I have a strong suspicion that something in Fedora's kernel sources or configuration is preventing it from working properly after the firmware update, but I have no idea where to start looking.

I have no idea either, but I'm currently having to run Fedora 42 with a OpenSUSE kernel. Have tried fedore nightly's and vanilla kernels from copr but they all exhibit the same behaviour. Having a quick look a the differences between fedora and opensuse there are hundreds of slight differences so without knowing the cause or how to debug it I have no idea what area of config might even be involved. It seems this also affects some XPS and Inspiron machines too.

pkgw, 2025-05-01:
No action on it yet, but I believe that [Fedora Bugzilla #2355276](https://bugzilla.redhat.com/show_bug.cgi?id=2355276) is the relevant ticket there.