fwupd/firmware-dell #216: OptiPlex 3080 and 5080 BIOS 1.30.1 pulled from Dell support site but still in fwupd?

pjwelsh, 2025-05-15:
@dell-client-linux 
**Hardware Information**
At least Dell OptiPlex 3080 and 5080 in fwupd.org has a *greater* version that the official Dell support website.

Here is the fwupd link that shows a 3080 2.30.1:
https://fwupd.org/lvfs/devices/com.dell.uefi10d19ee4.firmware

Here is the current latest 3080 Dell version of 2.29.0 still:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=gth4r&oscode=wt64a&productcode=optiplex-3080-sff

Here is the fwupd link that shows a 5080 1.30.1:
https://fwupd.org/lvfs/devices/com.dell.uefi459b329f.firmware

Here is the current latest 5080 Dell version of 1.29.0 still:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=v8xh8&oscode=wt64a&productcode=optiplex-5080-sff

So, did Dell remove a BIOS update that needs to be pulled in fwupd.org or did fwupd.org get an early release or????

As a note, I've not (yet?) noticed any issues with the fwupd.org provided 1.30.1 BIOS on the systems that have gotten this update.

tjtjan, 2025-05-18:
Noticing the same, still no sign off this new BIOS on dell's site.

pjwelsh, 2025-05-19:
Now I'm also noticing that the Optiplex 3090 is also having the same issue. The fwupd has a newer version (2.24.1 https://fwupd.org/lvfs/search?value=OptiPlex+3090) that does not exist on the official Dell support site 2.23.0 at https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=vkx4f&oscode=wt64a&productcode=optiplex-3090-sff

@dell-client-linux any info on why fwupd is getting non-official or recalled BIOS updates?

tjtjan, 2025-06-02:
Since Dell support is impossible to reach with an expired service tag, I just raised this issue via their bug bounty program. Hopefully they will answer there.

CragW, 2025-06-12:
The firmware versions on the Windows side have been upreved. They're a bit late in this case.

pjwelsh, 2025-06-12:
@CragW Thank you for the response on this.
I am a bit surprised that all the mentioned systems on *Dell's* site still show a lower version that is availible here, for *this* long (one month).
I'm hoping for an official notice/indicatoin that these BIOS updates are indeed "real production" and not either a supply chain infiltration nor a "bad" BIOS updated regressed from production.
@dell-client-linux

pjwelsh, 2025-06-12:
Thank you for pointing those out. I should have looked more closely... I just looked at the 3090 link and *it* is still not updated on the Dell web site:
https://www.dell.com/support/product-details/en-us/product/optiplex-3090-sff/drivers

The 3080 and 5080 do seem to have newer version now.

The crux of the issue still remains regarding the version mismatch and lack of Dell group acknowedgement of the issue(s) here.