fwupd/firmware-dell #217: Dell Inspiron 7610 fails to apply UEFI dbx update on reboot

enese94, 2025-06-09:
**Bug description:**

The UEFI Secure Boot Forbidden Signature Database (dbx) update fails to apply on a Dell Inspiron 16 Plus 7610 running Fedora 42. The fwupdmgr update command schedules the update correctly, but upon reboot, the UEFI firmware does not apply the capsule and boots straight into the OS. The fwupdmgr get-history command confirms the failure with the error: failed to run update on reboot: expected [new_version] and got [old_version].

This issue persists even after updating the main system BIOS to the latest version available on Dell's official support website.

**Affected Device**

    UEFI dbx

**Steps to Reproduce**

    On a Dell Inspiron 16 Plus 7610 with the latest system BIOS installed, run sudo fwupdmgr refresh.
    Run sudo fwupdmgr get-updates. The "Secure Boot dbx" update is shown.
    Run sudo fwupdmgr update to schedule the installation.
    Reboot the system as prompted.
    Observe: The system boots normally into Fedora without showing the UEFI firmware update screen.
    After logging in, run sudo fwupdmgr get-history to confirm the failure.

**Expected behavior**
The system should reboot into the UEFI capsule update environment (e.g., a blue screen or Dell screen) and apply the dbx update before booting into the main OS. fwupdmgr get-history should report the update as a "success".

**System Information**

    fwupd version: 
```
fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.11
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.9-300.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.11
```
    OS/Distribution: Fedora 42 (Workstation Edition)
    GNOME Version: 48.2
    Hardware Model: Dell Inspiron 16 Plus 7610
    Secure Boot: Enabled

Additional context
Please see the fwupdmgr get-history output below. It clearly shows the previous version (20230501) remains after the attempted update to 20241101.

```
sudo fwupdmgr get-history
Ponha o indicador direito sobre o leitor de impressões digitais
Dell Inc. Inspiron 16 7610
│
└─UEFI dbx:
  │   ID do dispositivo:  362301da643102b9f38477387e2193e57abaa590
  │   Versão anterior:    20230501
  │   Estado da atualização: Falhou
  │   Erro de atualização: failed to run update on reboot: expected 20241101 and got 20230501
  │   Última modificação: 2025-06-08
  │   GUID:               4a6cd2cb-8741-5257-9d1f-89a275dacca7
  │   Opções do dispositivo: • Dispositivo interno
  │                       • Atualizável
  │                       • Suporte no servidor remoto
  │                       • Precisa de uma reinício após a instalação
  │                       • A verificação criptográfica de hash está disponível
  │                       • O dispositivo pode ser usado durante a atualização
  │                       • Apenas atualizações de versão são permitidas
  │                       • Signed Payload
  │                       • Can tag for emulation
  │ 
  └─Atualização da configuração de Secure Boot dbx:
        Nova versão:      20241101
        ID remoto:        lvfs
        ID da versão:     105821
        Resumo:           UEFI Secure Boot Forbidden Signature Database
        Variação:         x64
        Licença:          Privativa
        Tamanho:          15,1 kB
        Criado:           2025-01-17
        Urgência:         Alta
        Fornecedor:       Linux Foundation
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Descrição:        
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Problemas:        529659
                          CVE-2024-7344
        Soma de verificação: d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f

```

raigoinabox, 2025-06-11:
I'm seeing a very similar error. Firmware update apparently succeeds but when I reboot the machine, it immediately boots into the OS and doesn't do any upgrade. Afterwards running `fwupdmgr get-history` will get this:

```
Dell Inc. Latitude 5540
│
└─System Firmware:
  │   Device ID:          71e8e3013ac4726cf6dd3476aaacd3047a25a3fc
  │   Previous version:   1.21.0
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot: expected 1.22.2 and got 1.21.0
  │   Last modified:      2025-06-11
  │   GUID:               24e6d799-c364-4c34-817e-020a080c29a9
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─Latitude 5540,Precision 3580 System Update:
        New version:      1.22.2
        Remote ID:        lvfs
        Release ID:       113024
        Summary:          Firmware for the Dell Latitude 5540,Precision 3580
        License:          Proprietary
        Size:             30,9 MB
        Created:          2025-05-14
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This stable release fixes the following issues:Firmware for the Dell Latitude 5540,Precision 3580
        Checksum:         545a2c0fa0bf74736643fa2adcfea9ee1a6c511241f5c0914fceead4298620e4
```

As you can see, it's a different machine. My OS is Ubuntu 25.04 and I used their Firmware application to run the update. I also mention that I have successfully upgraded my firmware before and this problem is new. Although I don't remember if I have successfully upgraded my firmware on Ubuntu 25.04 or an earlier version.

```
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.22
runtime   org.kernel                    6.14.0-15-generic
runtime   org.freedesktop.fwupd         2.0.7
```

hughsie, 2025-06-11:
@CragW wrong GUID?

CragW, 2025-06-12:
> wrong GUID?

Perhaps someone has encountered a similar issue on Ubuntu 25.04, see:
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2113748